pablozg/freeds #1: Fallo PWM manual

Tjchispas, 2020-03-08:
Hola, ante todo darte las gracias por el curro que te has pegado.
El problema que le encuentro es que al activar el PWM manualmente en la monitorización web del esp si aparece el porcentaje correcto, en 50% por mktt envía pwm=283 mi derivador esta conectado por mqtt y funciona de 0 a 100, no he podido probar con energía solar era demasiado tarde, en caso de que necesites mas datos o servir de betatester no dudes en contactar. Saludos

pablozg, 2020-03-08:
Hola, efectivamente era un fallo a la hora de convertir el valor de pwm para enviarlo por mqtt, bájate los binarios que acabo de subir y actualizas mediante la página web. No importa el orden, solamente tienes que esperar a que reinicie para actualizar el siguiente.

Tjchispas, 2020-03-08:
Ahora perfecto gracias, al 50% 750w de mi termo de 1500w, me gusta el modo de actualizar muy simple y efectivo.
Una petición para futuras versiones, puedes poner el configurar uno de los relés mqtt como un pwm mqtt, tengo un ESP8266 con tasmota configurado como dimmer y lo enlazo con el porcentaje de pwm mediante node red, pero hacerlo desde el FreeDs seria mucho mas fácil y podría usarlo mas gente.

pablozg, 2020-03-08:
No te entiendo con lo del relé pwm mqtt. ¿Puedes poner un ejemplo?

Respecto a lo del mqtt, el manual está puliendose, pero estos son los comandos por mqtt que puedes usar:

freeds/cmnd/pwm -> Acepta 1 para encender y 0 para apagar
freeds/cmnd/pwmman -> Acepta 1 para manual y 0 para auto
freeds/cmnd/pwmmanvalue -> Acepta valores de 0 a 100 que corresponde al % en modo manual
freeds/cmnd/screen -> Acepta valores de 0 a 4 corresponden a las pantallas de información en el oled
freeds/cmnd/pwmfrec -> Acepta valores entre 0 y 3000 corresponde a la frecuencia pwm
freeds/cmnd/brightness -> Acepta valores de 0 a 100 % corresponde al brillo de la pantalla

Tjchispas, 2020-03-08:
A cojonudo que ya tienes implementados los comandos, perfecto crearé algunos switch en HA

Le que no he sido capaz de explicarte es simplemente que uno de los 4 reles que de pueden configurar desde la web sea un pwm por mqtt con tasmota

Tjchispas, 2020-03-08:
[image]

pablozg, 2020-03-08:
Lo que quieres hacer es que cuando se active el relé envie un comando mqtt que tu elijas, en este caso uno para activar un switch en tasmota, no?

Tjchispas, 2020-03-08:
No exactamente lo que debe de enviar el el porcentaje pwm para que el tasmota configurado como dimmer actué sobre el dimmer físico, el esp32 no esta conectado físicamente al dimmer, por eso uso node red no encontré otro modo de enlazarlos, tampoco tengo los conocimientos suficientes

pablozg, 2020-03-08:
Vale, pero además de enviar el pwm con valor, el tema lo debes escribir tú, ya que si no lo indicas no hay forma de saber a quien enviarlo. 

Ok, lo tengo en cuenta y pensaré como integrarlo.

altusis1, 2020-03-09:
hola

altusis1, 2020-03-09:
hola pablozg. Muchas gracias por compartir esta maravilla de programa.
Tengo algunas dudas que si puedes me gustaria aclarases.
1-Tenéis algún grupo en telegram?
2-tengo sdm120 + Módulo TTL a RS485(xy-017)+ esp2866. Con lo que consigo que la info que aporta el sdm la paso a home assistant mediante mqtt. Puedo usar esto para conectarlo a freeds?
Un saludo muchas gracias por tu gran aporte.

vacarisses, 2020-03-15:
Hola,
el topic freeds/cmnd/pwmman a 0 para poner el PWM en AUTO no funciona, podrias poner un listado de topics.

Gracias

pablozg, 2020-03-15:
Gracias por el aviso, ya está solucionado, en cuanto pueda subo la nueva versión que será la 1.0.3.

Un poco más arriba puse el listado de comandos mqtt actuales, pero te los vuelvo a poner aquí, si creéis necesario alguno más indicarlo.

````
freeds/cmnd/pwm -> Acepta 1 para encender y 0 para apagar
freeds/cmnd/pwmman -> Acepta 1 para manual y 0 para auto
freeds/cmnd/pwmmanvalue -> Acepta valores de 0 a 100 que corresponde al % en modo manual
freeds/cmnd/screen -> Acepta valores de 0 a 4 corresponden a las pantallas de información en el oled
freeds/cmnd/pwmfrec -> Acepta valores entre 0 y 3000 corresponde a la frecuencia pwm
freeds/cmnd/brightness -> Acepta valores de 0 a 100 % corresponde al brillo de la pantalla
````

vacarisses, 2020-03-15:
Perfecto!!! Gracias, para mi con eso ya puedo automatizar en homeassistant, por otro lado, como puntalización ayer monte el ESP01 y en la web, no aparece que hay que puentearlo para que funcione, hasta que no lo hice no me funciono, solo como dato.

Gracias por tu proyecto y por tu ayuda

lito333, 2020-04-01:
Hola amigo gran trabajo.
Yo no puedo conectarlo porque el inversor que tengo es un easun power 5kw y no esta en la lista,pero tengo el icc solar que tiene mqtt.
mi pregunta es no se puede conectar mediante mqtt al icc solar,si es asi hay algun manual o alguna guia,
gracias y un saludo

closed 2020-05-02 by pablozg