pablozg/freeds #2: Echo de menos una introducción o manual

ffcadenas, 2020-03-08:
Gracias por tu trabajo pero así a pelo no sé ni por donde empezar y puede que esté orientado a otro público más instruido pero me gustaría aportar algo aunque sea como usuario. 
Qué aporta este software frente a otros. Es mas eficaz en gestionar excedentes o integrase con Home Assistant? 
Entiendo que es universal al integrar Shelly y wibeee.
Qué hardware necesita sólo el esp y el dimmer habitual o recomiendas alguno en concreto.
Gracias.

pablozg, 2020-03-08:
El manual lo estamos puliendo y en unos dos días estará listo.

Este programa es una mejora del original de Iqas, pero que debido a los numerosos cambios que se han realizado, se opto por crearlo como uno independiente, tiene todo lo que tenía el Iqas original pero más optimizado y mucho más configurable.

La diferencia sustancial es que admite el uso del meter directamente (con el uso de una placa conversora), por lo que no necesita acceso al inversor para poder hacer la gestión de los excedentes.

La inclusión del Shelly y el wibeee se ha añadido al ver las peticiones realizadas en el canal de telegram, pero no están probadas, por lo que si tienes la posibilidad de probar alguna y decirme si funciona te lo agradecería.

Sobre el hardware es compatible con los mismos que son compatibles con Iqas, aunque en un futuro se va a desarrollar un hardware más especifico y mejorado, la diferencia es que puedes elegir la frecuencia del pwm (0.5 Khz para el nuevo de aliexpress y 3.0 Khz para el Letón), y además dispone de una salida analógica por si alguien tiene alguna placa reguladora que necesite de una salida de ese estilo en lugar del pwm.

Como digo, cualquier tipo de sugerencia, mejora o prueba que puedas hacer, será bienvenida, por aquí o por el canal de telegram (Solares Chat o Solax Faq).

Gracias.

pablozg, 2020-03-11:
Ya está disponible el manual en http://freeds.es/

closed 2020-03-11 by pablozg

edu0017, 2020-05-22:
¿Alguien lo tiene con el meter independiente y la placa conversora?

edu0017, 2020-05-22:
¿Hay alguna instrucción para conectarlo a un wibeee?