pablozg/freeds #3: Vale, pero además de enviar el pwm con valor, el tema lo debes escribir tú, ya que si no lo indicas no hay forma de saber a quien enviarlo.

altusis1, 2020-03-09:
Vale, pero además de enviar el pwm con valor, el tema lo debes escribir tú, ya que si no lo indicas no hay forma de saber a quien enviarlo. 

Ok, lo tengo en cuenta y pensaré como integrarlo.

_Originally posted by @pablozg in https://github.com/pablozg/freeds/issues/1#issuecomment-596253783_

altusis1, 2020-03-09:
hola pablozg. Lo primero agradecerte en trabajo que estas haciendo pra que podamos utilizar esta maravilla de programa.
Si es posible me gustaria podrías solucionar una duda que tengo.
Tengo un sdm120 + un convertidor serie  a 485 + un esp 8266. Con ello lo que consigo es enviar los datos del sdm 120 medianto mqtt a home assistant. 
Puedo hacer que freeds recoja estos datos?

pablozg, 2020-03-09:
Hola, lo puedes hacer sin problema, sólo tienes que usar tasmota que tiene soporte para el sdm y mqtt, eliges la opción solax mqtt y configuras como tema del meter tudispositivo\tele\SENSOR, con eso debe conectarse y obtener datos del meter. También debes bajar la velocidad de publicación de tasmota a 10 segundos.

altusis1, 2020-03-09:
muchas gracias ahora lo pruebo

altusis1, 2020-03-09:
perdona no consigo que enlace.

Esta es mi configuración del sdm 120.
Host ()
192.168.2.50
Port (1883)
1883
Client (DVES_CAECB6)
DVES_%06X
User (DVES_USER)
test1
Password
••••
Topic = %topic% (sonoff)
SDM_120
Full Topic (%prefix%/%topic%/)
%prefix%/%topic%/

y esto lo que le envió a home assistan
   
19:32:30 MQT: tele/SDM_120/SENSOR = {"Time":"2020-03-09T19:32:30","ENERGY":{"Total":2180.327,"ActivePower":1203,"ApparentPower":1225,"ReactivePower":-246,"Frequency":50,"Factor":0.98,"Voltage":231,"Current":5.400,"PhaseAngle":0.00,"ImportActivePower":2142,"ExportActivePower":38,"ImportReactivePower":37,"ExportReactivePower":625,"TotalReactivePower":662}}

Configuración freeds:
monitorizacion - solax mqtt (tasmota)
configuración - 

DATOS MQTT
MQTT Activado
MQTT Broker:
192.168.2.50
MQTT Usuario:
test1
MQTT Password:
•••••
MQTT Puerto:
1883
Tema R1:
freeds/relay/1/STATUS
Tema R2:
freeds/relay/2/STATUS
Tema R3:
freeds/relay/3/STATUS
Tema R4:
freeds/relay/4/STATUS
Tema Solax:
tele/PZEM004T/SENSOR         **meter midiendo aporte solar**
Tema Meter:
tele/SDM_120/SENSOR           **meter midiendo entrada acometida**
CAMBIO DE PASSWORD
Clave de acceso:
Nueva clave de acceso:
API REMOTA
Web Api:

altusis1, 2020-03-09:
puede crear conflicto por estar enlazo a home assistant?

altusis1, 2020-03-09:
perdona otro dato importante no tengo todavía el esp32 con la pantalla y estoy utilizando uno normal sin pantalla el esp32 devkit v1

altusis1, 2020-03-09:
veo que me esta cogiendo la energía diaria solar del pzem. Aunque el valor aparece y desaparece.

Error en la recepción de los datos de medida
Monitorización
Potencia solar:
0.00 W
Potencia de red:
0.00 W
Energía diaria solar:
1.77 KWH 
Potencia string 1:
0.00 W (0.00 V) (0.00 A)
Potencia string 2:
0.00 W (0.00 V) (0.00 A)
PWM (AUTO):

que coincide con lo leido hoy por el pzem

PZEM004T_V3 Module
Pzem004T

Voltage | 231 V
Energy Today 0.030 A
Energy Today 0 W
Energy Today 7 VA
Power Factor 7 VAr
Power Factor 0.00
Power Factor 50 Hz
Energy Today  1.775 kWh
Power Factor 1.179 kWh
Power Factor 162.459 kWh

pablozg, 2020-03-10:
Buenos días, he subido la release 1.0.1 donde se corrige el fallo en la obtención de los datos mediante mqtt, pruébalo y me comentas.

altusis1, 2020-03-10:
Buenas tardes muchas gracias por tu dedicación y esfuerzo.
he subido los firmware.bin y spiffs.bin atravas de la web como indicas.
veo que la versión ahora es la 
Copyright © 2020. Derivador de energía solar excedente 1.0.1
Basado en openDS+ de Iqas, betatester: Aeizoon, (C) 2020 Pablo Zerón
Fecha compilación: Mar 10 2020 11:47:46

y después de ello sigo con el mismo problema.

Error en la recepción de los datos de medida
Monitorización
Potencia solar:
38.00 W
Potencia de red:
0.00 W
Energía diaria solar:
2.37 KWH
Potencia string 1:
0.00 W (0.00 V) (0.00 A)
Potencia string 2:
0.00 W (0.00 V) (0.00 A)
PWM (AUTO):

no recoge la lectura del sdm120. pero si la del pzem.
esta es la confi que tengo puesta.

Configuración
DATOS MQTT
MQTT Activado
MQTT Broker:
192.168.2.50
MQTT Usuario:
test1
MQTT Password:
•••••
MQTT Puerto:
1883
Tema R1:
freeds/relay/1/STATUS
Tema R2:
freeds/relay/2/STATUS
Tema R3:
freeds/relay/3/STATUS
Tema R4:
freeds/relay/4/STATUS
Tema Solax:
tele/PZEM004T/SENSOR
Tema Meter:
tele/SDM_120/SENSOR
CAMBIO DE PASSWORD
Clave de acceso:
Nueva clave de acceso:
API REMOTA
Web Api:
CONFIGURACIÓN PANTALLA OLED
Apagado autómatico de la pantalla desactivado
Tiempo para apagado automático (ms):
30000
CONFIGURACIÓN METER
Baud rate:
9600
Meter Id:
1
TIEMPO MÁXIMO DE ERROR
Tiempo máximo de error (ms):
20000
TIEMPO DE ADQUISICIÓN DE DATOS
Tiempo adquisición datos (ms):
1500
 
si invierto de esta manera

Tema Solax:
tele/SDM_120/SENSOR
Tema Meter:
tele/PZEM004T/SENSOR

aparece y desaparece el Error en la recepción de los datos de medida 
Potencia solar:
0.00 W
Potencia de red:
-18.00 W                  esta media es la producción de los paneles solares que es positiva POWER 18W pero la convierte en negativa 

Energía diaria solar:
0.00 KWH
Potencia string 1:
0.00 W (0.00 V) (0.00 A)
Potencia string 2:
0.00 W (0.00 V) (0.00 A)
PWM (AUTO):
17 %
SAL

no se donde puede estar el error ? 

Si necesitas que realice algún tipo de prueba me comentas.
Un saludo y gracias por tu tiempo y dedicación.

altusis1, 2020-03-10:
se me olvida , si dejo solo el pzem no monitoriza nada

Tema Solax:
tele/PZEM004T/SENSOR
Tema Meter:

si añado 
Tema Solax:
tele/PZEM004T/SENSOR
Tema Meter:
ddsu666/
o
tele/SDM_120/SENSOR

entonces recoge datos del pzem pero no quiere los del sdm120
Potencia solar:
10.00 W
Potencia de red:
0.00 W
Energía diaria solar:
2.38 KWH

pablozg, 2020-03-10:
Fijate en tu log de tasmota y en el mío:

Tu log:
````
tele/SDM_120/SENSOR = {"Time":"2020-03-09T19:32:30","ENERGY":{"Total":2180.327,"ActivePower":1203,"ApparentPower":1225,"ReactivePower":-246,"Frequency":50,"Factor":0.98,"Voltage":231,"Current":5.400,"PhaseAngle":0.00,"ImportActivePower":2142,"ExportActivePower":38,"ImportReactivePower":37,"ExportReactivePower":625,"TotalReactivePower":662}}
````

Mi log sdm120:
````
 Casa/PB/sdm120/tele/SENSOR = {"Time":"2020-03-10T19:18:20","ENERGY":{"TotalStartTime":"2020-03-06T10:10:01","Total":[2765.851,1951.085,2012.301],"Yesterday":13.198,"Today":9.491,"ExportActive":[551.061,185.302,365.759],"Period":2.93,"Power":367.50,"ApparentPower":391.41,"ReactivePower":139.30,"Factor":0.93,"Frequency":50.00,"Voltage":229.3,"Current":1.80,"ImportActive":2765.852,"ImportReactive":592.294,"ExportReactive":76.662,"PhaseAngle":0.00}}
````

Mi log ddsu666:
````
Casa/PB/ddsu666/tele/SENSOR = {"Time":"2020-03-10T19:21:27","ENERGY":{"TotalStartTime":"2020-03-09T00:00:00","Total":[2768.080,42310.609,3406.769],"Yesterday":4.660,"Today":3.860,"ExportActive":[953.010,662.020,290.990],"Period":0,"Power":429,"ApparentPower":475,"ReactivePower":161,"Factor":0.91,"Frequency":50,"Voltage":231,"Current":2.057}}
````

Si te das cuenta en los míos se reporta el valor de consumo de energía como "Power", que es el valor que leo en el derivador, por lo que creo que tienes una versión antigua de tasmota (yo tengo la 8.1.0.2), primero prueba a actualizarla y me comentas.

Respecto a que el valor desaparece es un acción programada, ya que una vez que supera el tiempo indicado sin leer el valor del meter resetea los valores a 0 y comienza de nuevo la cuenta.

Un saludo.

altusis1, 2020-03-10:
actualizado a la Tasmota 8.1.0 by Theo Arends y funciona lo tenea en la 6.6 . muchas gracias. 👍 fantástico, ............. que guay cundo funciona todo. bueno me queda cambiarlo en home assistant
 
Potencia solar:
0.00 W
Potencia de red:
-2302.00 W
Energía diaria solar:
2.38 KWH
Potencia string 1:
0.00 W (0.00 V) (0.00 A)
Potencia string 2:

o porque es negativa la potencia de red , no tendria que ser + puesto que entra de la acometida a casa

pablozg, 2020-03-10:
Viene heredado del firmware original de iqas y se ha mantenido por
compatibilidad y para no liar a las personas que usan opends+ y quieren
pasar a freeds.

altusis1, 2020-03-10:
ok muchas gracias por la aclaración

closed 2020-03-10 by pablozg