pablozg/freeds #4: Reinicios con inversor Fronius

usuariokike, 2020-03-10:
Hola, gracias de antemano por vuestro trabajo, estoy probado el gestor con un inversor Fronius y se reinicia constantemente, con el IQUAS no ocurre, no se si a alguien más le ocurre.

pablozg, 2020-03-10:
Hola, necesito más datos para valorar que puede ocurrir, necesito saber si sale algún mensaje en la Pantalla y si tienes opción de conectar el esp32 al ordenador y abrir un monitor serie (por ejemplo el que viene en el ide de arduino) con una velocidad de 115200bps para poder ver exactamente donde da el fallo.

Antes probaría si no lo has hecho ya, a borrar primero el esp32 antes de subir el nuevo firmware.

Cuando tengas el log del puerto serie lo añades en el mensaje y así puedo hacerme una idea de lo que pasa.

Gracias.

usuariokike, 2020-03-10:
Hola, es lo que he podido capturar del monitor serie.

CPU0 reset reason: Vbat power on reset
CPU1 reset reason: for APP CPU, reseted by PRO CPU

Testing EEPROM Library

EEPROM Size: 1028 bytes
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[WiFi-event] event: [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
2
0
Connecting to Wi-Fi...
[I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: HOME
[E][WiFiMulti.cpp:55] addAP(): [WIFI][APlistAdd] no ssid or ssid too long
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 1 - SCAN_DONE
[WiFi-event] event: 1
[I][WiFiMulti.cpp:114] run(): [WIFI] scan done
[I][WiFiMulti.cpp:119] run(): [WIFI] 2 networks found
[D][WiFiMulti.cpp:151] run():        0: [11][C0:25:E9:7B:4A:6A] Home_2 (-79) *
[D][WiFiMulti.cpp:149] run():  --->   1: [1][44:D4:37:3F:7A:39] HOME (-82) *
[I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: 44:D4:37:3F:7A:39 SSID: HOME Channal: 1 (-82)
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.1.159, MASK: 255.255.255.0, GW: 192.168.1.1
7
WiFi connected
IP address: 
192.168.1.159
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.1.159, MASK: 255.255.255.0, GW: 192.168.1.1
7
WiFi connected
IP address: 
192.168.1.159
[I][WiFiMulti.cpp:174] run(): [WIFI] Connecting done.
[D][WiFiMulti.cpp:175] run(): [WIFI] SSID: HOME
[D][WiFiMulti.cpp:176] run(): [WIFI] IP: 192.168.1.159
[D][WiFiMulti.cpp:177] run(): [WIFI] MAC: 44:D4:37:3F:7A:39
[D][WiFiMulti.cpp:178] run(): [WIFI] Channel: 1

WiFi connected
SSID: 
HOME
IP address: 
192.168.1.159
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 1 - SCAN_DONE
[WiFi-event] event: 1
HOME
Home_2
Welcome to FreeDS
Hostname: freeds_cbdc
mDNS responder started
[V][HTTPClient.cpp:235] beginInternal(): url: http://192.168.1.181/solar_api/v1/GetPowerFlowRealtimeData.fcgi
[D][HTTPClient.cpp:276] beginInternal(): host: 192.168.1.181 port: 80 url: /solar_api/v1/GetPowerFlowRealtimeData.fcgi
[D][HTTPClient.cpp:1025] connect():  connected to 192.168.1.181:80
[V][HTTPClient.cpp:1123] handleHeaderResponse(): RX: 'HTTP/1.1 200 OK'
[V][HTTPClient.cpp:1123] handleHeaderResponse(): RX: 'Content-Type: application/json'
[V][HTTPClient.cpp:1123] handleHeaderResponse(): RX: 'Cache-Control: no-cache, no-store, must-revalidate'
[V][HTTPClient.cpp:1123] handleHeaderResponse(): RX: 'Pragma: no-cache'
[V][HTTPClient.cpp:1123] handleHeaderResponse(): RX: 'Expires: 0'
[V][HTTPClient.cpp:1123] handleHeaderResponse(): RX: 'Transfer-Encoding: chunked'
[V][HTTPClient.cpp:1123] handleHeaderResponse(): RX: 'Date: Tue, 10 Mar 2020 18:27:52 GMT'
[V][HTTPClient.cpp:1123] handleHeaderResponse(): RX: 'Server: webserver'
[V][HTTPClient.cpp:1123] handleHeaderResponse(): RX: ''
[D][HTTPClient.cpp:1158] handleHeaderResponse(): code: 200
[D][HTTPClient.cpp:1165] handleHeaderResponse(): Transfer-Encoding: chunked
[D][HTTPClient.cpp:797] writeToStream():  read chunk len: 918
[D][HTTPClient.cpp:1295] writeToStreamDataBlock(): connection closed or file end (written: 918).
[D][HTTPClient.cpp:797] writeToStream():  read chunk len: 0
[D][HTTPClient.cpp:361] disconnect(): still data in buffer (2), clean up.
[... 384 lines cut ...]
Backtrace: 0x00000000:0x3ffb1da0 0x400f1170:0x3ffb1dc0 0x400f1241:0x3ffb1de0 0x400f2776:0x3ffb1e00 0x400dd5ed:0x3ffb1e40 0x400e218f:0x3ffb1ea0 0x400e2197:0x3ffb1ec0 0x400f2ab1:0x3ffb1ee0 0x400f2c5b:0x3ffb1f10 0x400df397:0x3ffb1f50 0x400f8be5:0x3ffb1fb0 0x40089501:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10044
load:0x40080400,len:5900
entry 0x400806ac

CPU0 reset reason: Software reset CPU
CPU1 reset reason: Software reset CPU

Testing EEPROM Library

EEPROM Size: 1028 bytes
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[WiFi-event] event: [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
2
0

pablozg, 2020-03-10:
Hola, puedes probar a flashear el esp con el firmware que he colgado aquí:

https://github.com/pablozg/freeds/releases/download/1.0.2_Fronius/firmware.bin

solo es sobreescribir el firmware.bin que tengas con este nuevo y flashear.

Si puedes vuelve a colgar el log con el resultado.

Gracias.

usuariokike, 2020-03-10:
Peor, los reinicios son más rápidos.
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10044
load:0x40080400,len:5900
entry 0x400806ac

CPU0 reset reason: Vbat power on reset
CPU1 reset reason: for APP CPU, reseted by PRO CPU

Testing EEPROM Library

EEPROM Size: 1028 bytes
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[WiFi-event] event: [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
2
0
Connecting to Wi-Fi...
[I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: HOME
[E][WiFiMulti.cpp:55] addAP(): [WIFI][APlistAdd] no ssid or ssid too long
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 1 - SCAN_DONE
[WiFi-event] event: 1
[I][WiFiMulti.cpp:114] run(): [WIFI] scan done
[I][WiFiMulti.cpp:119] run(): [WIFI] 3 networks found
[D][WiFiMulti.cpp:151] run():        0: [11][C0:25:E9:7B:4A:6A] Home_2 (-74) *
[D][WiFiMulti.cpp:149] run():  --->   1: [1][44:D4:37:3F:7A:39] HOME (-86) *
[D][WiFiMulti.cpp:151] run():        2: [7][9C:71:3A:A0:13:38] SAforWifi_RM (-91) *
[I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: 44:D4:37:3F:7A:39 SSID: HOME Channal: 1 (-86)
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 4 - STA_CONNECTED
[WiFi-event] event: 4
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.1.165, MASK: 255.255.255.0, GW: 192.168.1.1

WiFi connected
IP address: 
192.168.1.165
[I][WiFiMulti.cpp:174] run(): [WIFI] Connecting done.
[D][WiFiMulti.cpp:175] run(): [WIFI] SSID: HOME
[D][WiFiMulti.cpp:176] run(): [WIFI] IP: 192.168.1.165
[D][WiFiMulti.cpp:177] run(): [WIFI] MAC: 44:D4:37:3F:7A:39
[D][WiFiMulti.cpp:178] run(): [WIFI] Channel: 1

WiFi connected
SSID: 
HOME
IP address: 
192.168.1.165
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 1 - SCAN_DONE
[WiFi-event] event: 1
Home_2
HOME
SAforWifi_RM
Welcome to FreeDS
Hostname: freeds_cbdc
mDNS responder started
[V][HTTPClient.cpp:235] beginInternal(): url: http://192.168.1.181/solar_api/v1/GetPowerFlowRealtimeData.fcgi
[D][HTTPClient.cpp:276] beginInternal(): host: 192.168.1.181 port: 80 url: /solar_api/v1/GetPowerFlowRealtimeData.fcgi
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400e4c9c  PS      : 0x00060e30  A0      : 0x8019789a  A1      : 0x3ffb1d60  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3ffc1974  A5      : 0x00000001  
A6      : 0x00060420  A7      : 0x00000000  A8      : 0x80085479  A9      : 0x3ffb1d80  
A10     : 0x88fc2400  A11     : 0x3ffd0d40  A12     : 0x00060420  A13     : 0x00000001  
A14     : 0x00060423  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x88fc2408  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffff4  

Backtrace: 0x400e4c9c:0x3ffb1d60 0x40197897:0x3ffb1d90 0x400f2510:0x3ffb1db0 0x400f26bd:0x3ffb1de0 0x400f278f:0x3ffb1e20 0x400dd8a1:0x3ffb1e40 0x400e2403:0x3ffb1ea0 0x400e240b:0x3ffb1ec0 0x400f2e75:0x3ffb1ee0 0x400f301f:0x3ffb1f10 0x400df623:0x3ffb1f50 0x400f8f65:0x3ffb1fb0 0x40089501:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10044
load:0x40080400,len:5900
entry 0x400806ac

CPU0 reset reason: Software reset CPU
CPU1 reset reason: Software reset CPU

Testing EEPROM Library

EEPROM Size: 1028 bytes
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[WiFi-event] event: [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
2
0

pablozg, 2020-03-10:
Gracias por tu ayuda, ahora no estoy delante del ordenador, en un rato te vuelvo a subir un firmware nuevo.

pablozg, 2020-03-10:
Hola, ya he subido una nueva versión. Dime si mejora.

usuariokike, 2020-03-10:
Perdona, no la encuentro. Hay un enlace

pablozg, 2020-03-10:
https://github.com/pablozg/freeds/releases/download/1.0.2_Fronius/firmwaretest2.bin

iqas, 2020-03-10:
Si te sirve de ayuda, con el código que hay subido no reinicia (compilado por mi), con los binarios que subiste si lo hace, hasta con este último y no sólo con Fronius, en mi caso es general.

usuariokike, 2020-03-10:
Ahora parece que va bien.
Gracias

iqas, 2020-03-10:
Mas datos:
A los 5 min, se reinicia, te dejo log:

WM VALUE: % 84

PWM VALUE: 863

DATA_DISPLAY()

DATA_DISPLAY()

DATA_DISPLAY()

GETINVERTERDATA()
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x00000000  PS      : 0x00060230  A0      : 0x800ee2da  A1      : 0x3ffb1d40  
A2      : 0x3ffc17b8  A3      : 0x00000000  A4      : 0x3ffc26f8  A5      : 0x0070cbe5  
A6      : 0x3f40319d  A7      : 0x3f40319d  A8      : 0x8018c3de  A9      : 0x3ffb1d30  
A10     : 0x3ffd7330  A11     : 0x00000000  A12     : 0x0000002c  A13     : 0x3ffbda08  
A14     : 0x000f2da8  A15     : 0x00000002  SAR     : 0x00000010  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x00000000:0x3ffb1d40 0x400ee2d7:0x3ffb1d60 0x400ee349:0x3ffb1d80 0x400ef4cd:0x3ffb1da0 0x400db909:0x3ffb1de0 0x400dc454:0x3ffb1e90 0x400dc7b7:0x3ffb1ed0 0x400e2a61:0x3ffb1ef0 0x400e2c13:0x3ffb1f20 0x400dde49:0x3ffb1f60 0x400f4399:0x3ffb1fb0 0x400893d1:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10044
load:0x40080400,len:5900
entry 0x400806ac
Sonoff-4
WIFI1
WIFI2
INVITADOS

DATA_DISPLAY()

GETINVERTERDATA()
HTTPCODE ERROR: 200
JSON STRING: {
   "Body" : {

usuariokike, 2020-03-11:
Yo no he apreciado reinicios, pero si que la monitorización de los valores no funciona en la web.
tampoco en la 2º pantalla del esp32 lora

pablozg, 2020-03-11:
> Yo no he apreciado reinicios, pero si que la monitorización de los valores no funciona en la web.
> tampoco en la 2º pantalla del esp32 lora

Te mando un nuevo enlace por si lo puedes comprobar de nuevo (de esa página baja el firmwaretest3.bin), respecto a la web, si usas chrome prueba a forzar la recarga de página con ctrl+f5.

Si no te funciona descarga de la página anterior el spiffs.bin y actualizalo desde la web y vuelves a probar con el ctrl+f5 una vez se haya reiniciado.

Gracias.

pablozg, 2020-03-11:
@iqas Gracias por el log, despues de hacer varias pruebas el fallo proviene del timeout que le impuse a las conexiones http de 500ms, parece que el fronius tarda más tiempo de responder en algunas ocasiones y entonces fuerza el cierre de la conexión y produce el kernel panic.

La librería httpclient por defecto tiene un timeout de unos 15 segundos que unido a su funcionamiento síncrono, hace que se bloquee el correcto funcionamiento de todo el loop hasta que cierra la conexión (sobre todo con el inversor apagado o que no responde) por eso lo límite pero parece que tiene efectos secundarios.

Por eso he subido el timeout de todas las peticiones http a 4000ms para ver como responden, por mi parte lo he probado con el solax y tanto con 500ms como 4000ms funciona sin problemas.

En cuanto tenga algo más de tiempo, probaré a implementar las peticiones http de forma asíncrona por lo que ya nos tenemos que preocupar del timeout o de que no responda, y así ya todo el código trabajará de forma asíncrona.

Saludos.

usuariokike, 2020-03-11:
Gracias, sin reinicios y lee algunos datos del inversor en 2º página. Los valores de los string no los lee.
W,V,I de los String.

pablozg, 2020-03-11:
> Gracias, sin reinicios y lee algunos datos del inversor en 2º página. Los valores de los string no los lee.
> W,V,I de los String.

Eso es debido a la api de Fronius que no envía esos datos.

usuariokike, 2020-03-11:
ok, gracias

closed 2020-03-14 by pablozg