pablozg/freeds #6: Conexión con icc solar

pablozg, 2020-04-02:
Hola amigo gran trabajo.
Yo no puedo conectarlo porque el inversor que tengo es un easun power 5kw y no esta en la lista,pero tengo el icc solar que tiene mqtt.
mi pregunta es no se puede conectar mediante mqtt al icc solar,si es asi hay algun manual o alguna guia,
gracias y un saludo

_Originally posted by @lito333 in https://github.com/pablozg/freeds/issues/1#issuecomment-607086087_

pablozg, 2020-04-02:
@lito333 para poder usarlo con iccsolar necesito que me copies aquí la información que envía iccsolar mediante mqtt, porque creo que es una funcionalidad extra de iccsolar y no sé si la tienes activa.

Por el momento solo es compatible con tasmota mediante mqtt, pero si me envías la cadena json que publica iccsolar puedo implementarlo.

lito333, 2020-04-02:
[{"id":"36","userid":"1","name":"node:1:batterywattskwhd","datatype":"2","tag":"Node 1","public":"1","size":"72","engine":"2","processList":"","time":"1585862360","value":"-0.14249238888934"},{"id":"38","userid":"1","name":"node:1:loadwattskwhd","datatype":"2","tag":"Node 1","public":"1","size":"72","engine":"2","processList":"","time":"1585862360","value":"5.3908241666648"},{"id":"39","userid":"1","name":"node:1:pvwattskwhd","datatype":"2","tag":"Node 1","public":"1","size":"72","engine":"2","processList":"","time":"1585862360","value":"5.1322005555712"},{"id":"40","userid":"1","name":"node:1:gridwattskwhd","datatype":"2","tag":"Node 1","public":"1","size":"72","engine":"2","processList":"","time":"1585862360","value":"1.6792617777765"},{"id":"42","userid":"1","name":"node:1:loadwatts","datatype":"1","tag":"Node 1","public":"1","size":"0","engine":"5","time":"1585862360","value":"195"},{"id":"43","userid":"1","name":"node:1:gridvoltage","datatype":"1","tag":"Node 1","public":"1","size":"0","engine":"5","time":"1585862360","value":"235.3"},{"id":"44","userid":"1","name":"node:1:inverter_voltage","datatype":"1","tag":"Node 1","public":"1","size":"0","engine":"5","time":"1585862360","value":"235.3"},{"id":"45","userid":"1","name":"node:1:pvwatts","datatype":"1","tag":"Node 1","public":"1","size":"0","engine":"5","time":"1585862360","value":"0"},{"id":"46","userid":"1","name":"node:1:pvwatts1","datatype":"1","tag":"Node 1","public":"1","size":"0","engine":"5","time":"1585862360","value":"0"},{"id":"47","userid":"1","name":"node:1:pvwatts2","datatype":"1","tag":"Node 1","public":"1","size":"0","engine":"5","value":"0","time":"1585862360"},{"id":"48","userid":"1","name":"node:1:pvVolts1","datatype":"1","tag":"Node 1","public":"1","size":"0","engine":"5","time":"1585862360","value":"0"},{"id":"49","userid":"1","name":"node:1:pvAmps1","datatype":"1","tag":"Node 1","public":"1","size":"0","engine":"5","time":"1585862360","value":"0"},{"id":"50","userid":"1","name":"node:1:pvVolts2","datatype":"1","tag":"Node 1","public":"1","size":"0","engine":"5","time":"1585862360","value":"0"},{"id":"51","userid":"1","name":"node:1:pvAmps2","datatype":"1","tag":"Node 1","public":"1","size":"0","engine":"5","time":"1585862360","value":"0"},{"id":"52","userid":"1","name":"node:1:loadpercentage","datatype":"1","tag":"Node 1","public":"1","size":"0","engine":"5","time":"1585862360","value":"4"},{"id":"53","userid":"1","name":"node:1:inverter_temp","datatype":"1","tag":"Node 1","public":"1","size":"0","engine":"5","time":"1585862360","value":"47"},{"id":"54","userid":"1","name":"node:1:gridwatts","datatype":"1","tag":"Node 1","public":"1","size":"0","engine":"5","time":"1585862360","value":"195"},{"id":"55","userid":"1","name":"node:1:soc","datatype":"1","tag":"Node 1","public":"1","size":"0","engine":"5","time":"1585862360","value":"14"},{"id":"56","userid":"1","name":"node:1:batteryvolts","datatype":"1","tag":"Node 1","public":"1","size":"0","engine":"5","time":"1585862360","value":"44.6"},{"id":"57","userid":"1","name":"node:1:SolarWatts","datatype":"1","tag":"Node 1","public":"1","size":"0","engine":"5","time":"1585862360","value":"0"},{"id":"58","userid":"1","name":"node:1:batteryamps","datatype":"1","tag":"Node 1","public":"1","size":"0","engine":"5","time":"1585862360","value":"-1"},{"id":"59","userid":"1","name":"node:1:grid_frequency","datatype":"1","tag":"Node 1","public":"1","size":"0","engine":"5","time":"1585862360","value":"50"},{"id":"60","userid":"1","name":"node:1:inverter_frequency","datatype":"1","tag":"Node 1","public":"1","size":"0","engine":"5","time":"1585862360","value":"50"},{"id":"61","userid":"1","name":"node:1:batterywatts","datatype":"1","tag":"Node 1","public":"1","size":"0","engine":"5","time":"1585862360","value":"-44.6"},{"id":"62","userid":"1","name":"node:1:efficiency","datatype":"2","tag":"Node 1","public":"1","size":"0","engine":"2","time":"1585862360","value":"0.029570597221903"},{"id":"63","userid":"1","name":"node:1:solarmode","datatype":"1","tag":"Node 1","public":"1","size":"0","engine":"5","time":"1585862360","value":"0"},{"id":"64","userid":"1","name":"node:1:PI_Temperature","datatype":"1","tag":"Node 1","public":"1","size":"0","engine":"5","time":"1585862360","value":"28.4"},{"id":"65","userid":"1","name":"node:1:Max_Load_Hour","datatype":"1","tag":"Node 1","public":"1","size":"0","engine":"5","time":"1585862360","value":"232"},{"id":"66","userid":"1","name":"node:1:Max_Load_Day","datatype":"1","tag":"Node 1","public":"1","size":"0","engine":"5","time":"1585862360","value":"2101"},{"id":"67","userid":"1","name":"node:1:Max_PV_Day","datatype":"1","tag":"Node 1","public":"1","size":"0","engine":"5","time":"1585862360","value":"1221"},{"id":"68","userid":"1","name":"node:1:Max_PV_Hour","datatype":"1","tag":"Node 1","public":"1","size":"0","engine":"5","value":"0","time":"1585862360"},{"id":"69","userid":"1","name":"node:1:Grid_KWh_Used","datatype":"1","tag":"Node 1","public":"1","size":"0","engine":"5","time":"1585862360","value":"2.044"},{"id":"70","userid":"1","name":"node:1:PV_KWh_Produced","datatype":"1","tag":"Node 1","public":"1","size":"0","engine":"5","time":"1585862360","value":"5.121"},{"id":"71","userid":"1","name":"node:1:Total_KWh_Used","datatype":"1","tag":"Node 1","public":"1","size":"0","engine":"5","time":"1585862360","value":"5.748"},{"id":"72","userid":"1","name":"node:1:ICC_Hour_Runtime","datatype":"1","tag":"Node 1","public":"1","size":"0","engine":"5","time":"1585862360","value":"51"},{"id":"73","userid":"1","name":"node:1:ICC_Minute_Runtime","datatype":"1","tag":"Node 1","public":"1","size":"0","engine":"5","value":"45","time":"1585862360"},{"id":"74","userid":"1","name":"node:1:ICC_Sec_Runtime","datatype":"1","tag":"Node 1","public":"1","size":"0","engine":"5","time":"1585862360","value":"52"}]

lito333, 2020-04-02:
Haber si esto porque no se me da muy bien la informatica

pablozg, 2020-04-03:
> Haber si esto porque no se me da muy bien la informatica

Gracias por la info, necesito también que me digas si el tema en el que publica esa información es fijo o lo podéis cambiar en la configuración del icc solar, en caso de ser fijo necesito que me digas cual es el tema para poder añadirlo.

lito333, 2020-04-03:
es algo de esto

 Admite el protocolo MQTT (TCP) y Websocket (especialmente útil para redes restringidas de cortafuegos). 
SSL para la comunicación segura. 
Soporte JSON tanto para suscribirse como para publicar mensajes. 
 Los paneles se suscriben y / o publican el tema automáticamente, por lo tanto, actualícese en tiempo real. 
Diseñado para trabajar con un agente público de manera eficiente (utilizando el prefijo del dispositivo). 
 Marca de tiempo enviada y recibida del agente.
Escribir clave API
fc30c66f4b96c30a3f222d8635b017f0
Leer clave API
6285e375066bf9d215a0efa5b9481f54
Subscribe Ack Received for MsgID : 230
 o lo que necesitas son los topics

pablozg, 2020-04-03:
@lito333 He estado haciendo pruebas con la cadena json que me has enviado, pero es demasiado pesada para procesarla correctamente con el esp32, he visto en el manual del icc que existen diversos topics para mqtt

````

Inverter/Name
Inverter/LoadWatts
Inverter/InverterVolts
Inverter/InverterFreq
Inverter/LoadPercentage
Inverter/BatteryVolts
Inverter/BatteryAmps
Inverter/BatteryWatts
Inverter/BatterySOC
Inverter/Temperature
Inverter/InverterMode
Inverter/Off_Grid_Time
Inverter/RPI_Temperature
Inverter/GridWatts
Inverter/GridVoltage
Inverter/GridFreq
Inverter/PvWattsTotal
Inverter/Inverter2_MPPT1_current
Inverter/Inverter2_MPPT2_Current
Inverter/Inverter2_MPPT3_Current
Inverter/Inverter2_MPPT1_Volts
Inverter/Inverter2_MPPT2_Volts
Inverter/Inverter2_MPPT2_Volts
Inverter/Inverter2_MPPT1_Watts
Inverter/Inverter2_MPPT2_Watts
Inverter/Inverter2_MPPT3_Watts
Inverter/MPPT1_Watts
Inverter/MPPT2_Watts
Inverter/MPPT3_Watts
Inverter/MPPT1_Volts
Inverter/MPPT2_Volts
Inverter/MPPT3_Volts
Inverter/MPPT1_Amps
Inverter/MPPT2_Amps
Inverter/MPPT3_Amps
Inverter/GridKwUse
Inverter/TotalKwUse
Inverter/SolarKwUse
Inverter/MaxLoadDay
Inverter/MaxPvDay
Inverter/Solar_Efficiency
Inverter/ICC_Runtime
Inverter/ICC_Version
Pylontech/Cycles
Pylontech/Amps
Pylontech/Volts
Pylontech/Temperature
Pylontech/Remaining_AH
Pylontech/SOC
Inverter/Month_Grid_Total
Inverter/Month_Total_Use
Inverter/Month_PV_Total
Inverter/Month_Cost_Saving
Inverter/Month_TotalCost
BMV/DepthOfLastDischarge
BMV/AH_Consumed
BMV/Watts
BMV/Amps
BMV/Voltage
BMV/MidpointVoltage
BMV/Midpoint
BMV/MidpointDeviation
BMV/ChargeEnergy
BMV/DischargeEnergy
BMV/Efficiency
BMV/DepthOfDeepestDischarge
BMV/DepthOfAverageDischarge
BMV/NumberOfChargeCycles
BMV/NumberOfFullDischarges
BMV/TimeSinceLastFullCharge
BMV/SOC
BMV/TimeToGo
BMV/TimeTo100
````
Me puedes confirmar sin son válidos en tu caso? ya que así es más sencillo procesar los datos correctamente, leyendo únicamente los valores necesarios.

¿Puedes enviarme una captura con los datos que envían esos topics? Gracias.

pablozg, 2020-04-03:
> es algo de esto
>
> Admite el protocolo MQTT (TCP) y Websocket (especialmente útil para redes restringidas de cortafuegos).
> SSL para la comunicación segura.
> Soporte JSON tanto para suscribirse como [...]

Si, necesito saber de que topic obtienes el json que me has pasado y si los topics que te he puesto en el mensaje anterior son válidos en tu instalación.

lito333, 2020-04-03:
lo saque de aqui
API de feeds
Autenticación Apikey
Si desea llamar a cualquiera de las siguientes acciones cuando no ha iniciado sesión, tiene estas opciones para autenticarse con la clave API:

Agregue en la URL de su solicitud: & apikey = APIKEY
Utilice el parámetro POST: "apikey = APIKEY"
Agregue el encabezado HTTP: "Autorización: Portador APIKEY"
Solo lectura:
6285e375066bf9d215a0efa5b9481f54

Leer escribir:
fc30c66f4b96c30a3f222d8635b017f0

HTML
http: // localhost / emoncms / feed / list - La vista de lista de fuentes

http: // localhost / emoncms / feed / api - Esta página

JSON
Para usar la API json, la url de solicitud debe incluir .json

Leer acciones de feed

Listar feeds para usuarios autenticados	http: //localhost/emoncms/feed/list.json
Listar feeds públicos para el usuario dado	http: //localhost/emoncms/feed/list.json? userid = 0
Obtener campo de alimentación	http: //localhost/emoncms/feed/get.json? id = 1 & field = name
Obtenga todos los campos de alimentación	http: //localhost/emoncms/feed/aget.json? id = 1
Obtener meta meta (PHPFina, PHPFiwa)	http: //localhost/emoncms/feed/getmeta.json? id = 1
Leer acciones de datos de feed

Última hora y valor actualizados para el feed	http: //localhost/emoncms/feed/timevalue.json? id = 1
Último valor de un feed dado	http: //localhost/emoncms/feed/value.json? id = 1
Último valor para múltiples feeds	http: //localhost/emoncms/feed/fetch.json? ids = 1,2,3
Devuelve datos de feed	http: //localhost/emoncms/feed/data.json? id = 0 & start = UNIXTIME & end = UNIXTIME & interval = 5
Devuelve datos de histograma	http: //localhost/emoncms/feed/histogram.json? id = 0 & start = UNIXTIME & end = UNIXTIME
Devuelve kwh consumido en una banda de potencia dada usando el tipo de datos de histograma	http: //localhost/emoncms/feed/kwhatpower.json? id = 0 & min = & max =
Exportar datos CSV (formato de tiempo = 1: proporciona formato de cadena de fecha y hora)	http: //localhost/emoncms/feed/csvexport.json? id = 0 & start = UNIXTIME & end = UNIXTIME & interval = 60 & timeformat = 1 =
Escribir acciones de datos de feed

Insertar nuevo punto de datos	http: //localhost/emoncms/feed/insert.json? id = 0 & time = UNIXTIME & value = 100.0
Actualizar punto de datos	http: //localhost/emoncms/feed/update.json? id = 0 & time = UNIXTIME & value = 100.0
Eliminar punto de datos	http: //localhost/emoncms/feed/deletedatapoint.json? id = 0 & feedtime = UNIXTIME
Acciones de configuración de feed

Crear nuevo feed	http: //localhost/emoncms/feed/create.json? tag = Test & name = Power & datatype = 1 & engine = 5 & options = {"intervalo": 10}
Eliminar feed existente	http: //localhost/emoncms/feed/delete.json? id = 0
Actualizar campo de feed	http: //localhost/emoncms/feed/set.json? id = 0 & fields = {'name': 'anewname'}
Devolver el tamaño total de los motores	http: //localhost/emoncms/feed/updatesize.json
Devolver puntos de almacenamiento intermedio pendientes de escritura	http: //localhost/emoncms/feed/buffersize.json
Acciones del proceso de alimentación virtual

Obtener lista de proceso de alimentación	http: //localhost/emoncms/feed/process/get.json? id = 1
Establecer lista de proceso de alimentación	http: //localhost/emoncms/feed/process/set.json? id = 0 & processlist = 0: 0
Restablecer lista de proceso de alimentación	http: //localhost/emoncms/feed/process/reset.json? id = 0
Desarrollado por OpenEnergyMonitor.org | 9.8.30 | 2018.05.08

pablozg, 2020-04-03:
Entonces si escribes esto en el navegador, que te sale? Copia lo que responde por cada una de las lineas.

http: //ip_del_iccsolar/emoncms/feed/list.json?userid=0
http: //ip_del_iccsolar/emoncms/feed/list.json?userid=1
http: //ip_del_iccsolar/emoncms/feed/aget.json?id=1

lito333, 2020-04-03:
http: //ip_del_iccsolar/emoncms/feed/list.json?userid=0
[]
http: //ip_del_iccsolar/emoncms/feed/list.json?userid=1

lito333, 2020-04-03:
http: //ip_del_iccsolar/emoncms/feed/list.json?userid=0
[]
http: //ip_del_iccsolar/emoncms/feed/list.json?userid=1
nada
http: //ip_del_iccsolar/emoncms/feed/aget.json?id=1
{"success":false,"message":"Feed does not exist"}

lito333, 2020-04-03:
encontre esto 
JSON format:	curl --data "data={power1:100,power2:200,power3:300}" "http://localhost/emoncms/input/post.json?node=1&apikey=fc30c66f4b96c30a3f222d8635b017f0"

lito333, 2020-04-03:
For example using the first json type request above just add the apikey to the end like this:	http://localhost/emoncms/input/post.json?node=1&json={power:200}&apikey=fc30c66f4b96c30a3f222d8635b017f0

pablozg, 2020-04-04:
Hola, 

Baja esta versión y prueba a ver que tal te va. https://github.com/pablozg/freeds/releases/download/Pre1.0.5/FreeDs_Pre1.0.5.zip

Las instrucciones para programarlo las tienes en www.freeds.es

Necesitaría también que me indiques que valores te dan todos los temas que puse arriba para saber exactamente de donde extraer los datos. (Sobre todo la producción solar diaria)

lito333, 2020-04-04:
hola acavo de instalarlo y funciona me da sola la potencia de red pero como es de noche no hay otros datos,mañana de dia lo pruebo haber si salen el resto de datos

lito333, 2020-04-04:
[cid:image003.png@01D60ACF.DB742BA0]

lito333, 2020-04-04:
Inverter/Name
Inverter/LoadWatts...........................................salida del inversor en watios 200watios
Inverter/InverterVolts......................................salida del inversor en voltios 230v
Inverter/InverterFreq.......................................salida del inversor frecuencia 50mhz
Inverter/LoadPercentage...............................porcentaje al que esta trabajando el inversor 20%
Inverter/BatteryVolts........................................voltaje de la bateria 48v
Inverter/BatteryAmps......................................Amperios que recive (carga) 20A o da (descarga)-20A la bateria
Inverter/BatteryWatts......................................Watios que recive (carga) 200W o da (descarga)-200W la bateria
Inverter/BatterySOC........................................% de carga de la bateria 80%
Inverter/Temperature......................................temperatura del inversor 40º
Inverter/InverterMode...................................modo en el que esta trabajando el inversor(utility,SBU,solar first)
Inverter/Off_Grid_Time
Inverter/RPI_Temperature...........................temperatura rasberry pi 30º
Inverter/GridWatts.........................................Watios que esta consumiendo el inversor de la red 200W
Inverter/GridVoltage....................................voltaje de la red 230V
Inverter/GridFreq..........................................frecuencia de la red 50mhz
Inverter/PvWattsTotal.................................Watios que estan generando las placas solares 800W
Inverter/Inverter2_MPPT1_current
Inverter/Inverter2_MPPT2_Current
Inverter/Inverter2_MPPT3_Current
Inverter/Inverter2_MPPT1_Volts
Inverter/Inverter2_MPPT2_Volts----------------estos son para dos inversores en paralelo
Inverter/Inverter2_MPPT2_Volts
Inverter/Inverter2_MPPT1_Watts
Inverter/Inverter2_MPPT2_Watts
Inverter/Inverter2_MPPT3_Watts
Inverter/MPPT1_Watts....................1º strem watios que estan generando las placas  800W
Inverter/MPPT2_Watts
Inverter/MPPT3_Watts
Inverter/MPPT1_Volts........................1º strem voltaje que estan generando las placas  120V
Inverter/MPPT2_Volts
Inverter/MPPT3_Volts
Inverter/MPPT1_Amps....................1º stren Amperaje que estan generando las placas 80A
Inverter/MPPT2_Amps
Inverter/MPPT3_Amps
Inverter/GridKwUse.......................Kwh diarios usados de la red 1.86kwh
Inverter/TotalKwUse.....................Kwh diarios que han salido del inversor 5.22kwh
Inverter/SolarKwUse.....................Kwh  solares diarios usados de la 4.77kwh
Inverter/MaxLoadDay.................carga (pico)mas alta del dia en kw 2381W
Inverter/MaxPvDay......................pico mas alto en W de la generacion solar 1213W
Inverter/Solar_Efficiency............eficiencia del sistema solar 3.18 kwh/kw
Inverter/ICC_Runtime
Inverter/ICC_Version..................version del sof del icc 2.995
Pylontech/Cycles
Pylontech/Amps
Pylontech/Volts --------------------------------estos son para bateria pylontech
Pylontech/Temperature
Pylontech/Remaining_AH
Pylontech/SOC
Inverter/Month_Grid_Total
Inverter/Month_Total_Use
Inverter/Month_PV_Total
Inverter/Month_Cost_Saving
Inverter/Month_TotalCost
BMV/DepthOfLastDischarge
BMV/AH_Consumed
BMV/Watts
BMV/Amps
BMV/Voltage
BMV/MidpointVoltage
BMV/Midpoint
BMV/MidpointDeviation
BMV/ChargeEnergy----------------------estos son para un medidor de corriente de bateria BMV
BMV/DischargeEnergy
BMV/Efficiency
BMV/DepthOfDeepestDischarge
BMV/DepthOfAverageDischarge
BMV/NumberOfChargeCycles
BMV/NumberOfFullDischarges
BMV/TimeSinceLastFullCharge
BMV/SOC
BMV/TimeToGo
BMV/TimeTo100

lito333, 2020-04-05:
Hola buenos dias,no hay mucha produccion solar que llueve.
Hay un error en las mediciones algunos datos marcan de mas
Potencia solar:  1000.00 W-----------  valor real(100.00W)
Potencia de red: 193.00 W-----------esta ok valor real (193.00W)
Energía diaria solar: 0.00 KWH------no hay datos valor real (105.00W)
Potencia string 1: 1000.00 W (870.00 V) (2.00 A)----valor real(100.00W )(87.00V)(2.00A)
Potencia string 2: 0.00 W (0.00 V) (0.00 A)
PWM (AUTO): 100 %

cada 8 10 0 20 segundos el freeds saca un mensaje :
Error en la recepcion de los datos de medida se ponen los datos a cero y en un segundo vuelven a salir los datos

pablozg, 2020-04-05:
Hola, Actualiza el firmware.bin usando esta nueva versión y me dices:
https://github.com/pablozg/freeds/releases/download/1.0.5/FreeDs_Pre1.0.5rev2.zip

lito333, 2020-04-05:
Hola pablo,ahora los valores son todos correctos
lo unico que me dejo de funcionar es el PWM en automatico
en manual si funciona.
Sale muy amenudo el error:
Error en la recepción de los datos de medida
y cuando sale el error los datos se ponen todos a cero un momento
voi a conectarlo al calentador que lo saque para actualizarlo y te digo como va

https://github.com/pablozg/freeds/releases/download/1.0.5/FreeDs_Pre1.0.5rev2.zip

pablozg, 2020-04-05:
Yo lo he estado probando con otro compañero que lo tiene y funciona sin problemas, comprueba que tengas buena cobertura de la wifi y aumenta el Tiempo máximo de error

closed 2020-04-13 by pablozg