pablozg/freeds #8: fallo freeds

lito333, 2020-04-22:
Buenos dias pablo,no consigo que me funcione el freeds con el icc estoi arto de darle vueltas y no me va 
la pantalla marca bien los valores tanto de red como de produccion solar,y el pwm si lo activo yo manualmente el calentador si me funciona hasta el porcentaje que yo le ponga si le pongo un 50% pues calienta a un 50% si le pongo el 80% va hasta el 80%,
pero con el pwm  en automatico no enciende nunca el calentador aunque tenga una produccion solar de 2500w,
si cuando lo pongo el pwm en manual y esta funcionando el calentador,si lo paso a automatico va bajando el porcentaje hasta caer a cero y se apaga el calentadro.
si tienes alguna idea o puedes ponerme en contacto con algien que lo tenga funcionando con el icc 
para encontrar donde puedo tener el fallo,sino ya lo dejo por imposible porque ya no se que hacerle
un saludo

pablozg, 2020-04-22:
Antes de tirar la toalla, ¿estás en alguno de los grupos de Telegram (Solares o SolaxX1)?, si es así contacta conmigo, pero necesito que me indiques con que signo te aparece en la página de monitorización la potencia de red cuando estás vertiendo a red, porque puede que sea el problema.

Si te aparece con signo negativo cuando en realidad estás vertiendo es un problema del software que se soluciona en un segundo,

Un saludo.

lito333, 2020-04-22:
Si que te hablo por privado o por publico

closed 2020-05-02 by pablozg