pablozg/freeds #9: Modbus Victron

avm162, 2020-05-13:
Hola Pablo,
No se si lo tienes reportado. Al usar los datos Modbus del Victron, me da un error en el Unit id 288 en la instancia 239( en errores del modbus del del CCGX. Gran trabajo. 
Gracias.

pablozg, 2020-05-13:
Hola, lo primero necesito que actualices a la última versión, tienes que borrarlo y graba todo de nuevo, ya que han habido numerosos cambios. 

Cuando lo hayas hecho dime si te sigue saliendo el error.

Gracias.

https://github.com/pablozg/freeds/releases/download/1.0.5/FreeDS_Pre1.0.5Rev7.zip

avm162, 2020-05-13:
Sigue igual. De hecho el id 239 corresponde al VE.Direct por usb. Normalmente se consulta los valores principales en el id 100. No se si ahí puede estar el fallo.

pablozg, 2020-05-13:
Hola, el victron está así porque otro usuario estuvo haciendo pruebas y lo dejamos funcionando con esos valores.

Del id 239 obtengo:
 Frecuencia (red), potencia (red), voltaje (batería), intensidad (batería) y temperatura.

y del id 100 la potencia de las baterías.

Mándame un privado por alguno de los grupos de telegram de solares o solax. y te mando una versión sacando los datos desde el id 100, ¿puedes enviarme que instancias tienes disponibles en tu victron?

metragi, 2020-05-20:
Hola, yo tengo el mismo problema que avm162. No tengo ninguno de estos grupos de telegrama pero si me dejas enlace te mando la lista de ids que permite de un Excel de victron y las instancias que tengo disponibles, que también tengo un Carlo gavazzi. Gracias y un saludo

pablozg, 2020-05-20:
> Hola, yo tengo el mismo problema que avm162. No tengo ninguno de estos grupos de telegrama pero si me dejas enlace te mando la lista de ids que permite de un Excel de victron y las instancias que tengo disponibles, que también tengo un Carlo gavazzi. Gracias y un saludo

Hola, en principio este problema ya está solucionado, en breve sacaré de nuevo la revisión 8 donde estará funcionado. Si aún así tuvieras alguna pega me lo dices.

Gracias

closed 2020-05-30 by pablozg