pablozg/freeds #10: Medidor de corriente con la sonda SCT013

rdaguerre, 2020-07-23:
Antes que nada te agradezco este proyecto. Te quedó muy nuevo presentado y todo. Saludos
Mi aporte sería que a veces se tiene la placa del freeds cerca de la entrada de luz. Y tal vez se podría implementar que usando una SCT013 de pueda leer la corriente y no precise los medidores modbus.
Saludos y te felicito

pablozg, 2020-07-29:
Perdón por el retraso en la contestación, las sondas las tengo y en cuanto pueda quiero hacer algunas pruebas, pero por el momento no hay fecha para ello.

nvvee, 2020-09-14:
Justo hoy estaba mirando lo mismo. Las sondas SCT013 estan bien y no son invasivas pero implican tener que disenyar un circuito (resistencia + condensador, usando una placa de prototipos o hacer una pequeña PCB, etc). Estaria bien si la solución se pudiera adaptar a cosas más "comerciales" como https://www.amazon.es/electronics-salon-montaje-150amp-Corriente-Sensor/dp/B016M63GTM/ref=sr_1_32?dchild=1&keywords=sensor+corriente+arduino&qid=1600076706&sr=8-32 o https://es.aliexpress.com/item/32951258792.html?spm=a2g0o.productlist.0.0.6b8e2e0bKOMppb&algo_pvid=005b53f3-e9e2-4fc8-9f87-6226674d5dd0&algo_expid=005b53f3-e9e2-4fc8-9f87-6226674d5dd0-2&btsid=0b0a050b16000775763576651e7edf&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_ con un precio relativamente bajo (5 euros con envio en aliexpress) haciendo la solucion muy accesible a cualquier perfil aunque no sea tecnico.

Dicho esto, si pasa a usarse cualquiera de estas soluciones con medidor de corriente en la carga del termo, seria posible también usar estos datos para hacer un cálculo (aproximado) de los kwh consumidos por la carga? En principio solo seria qüestión de sumar los consumos instantanios de la carga siempre que se detecte que hay salida de corriente hacia la carga (ya que los que no usamos sensor de temperatura tenemos el pwm al 100% pero la carga desconectada y sin consumir realmente debido a la desconnexión a través del termostato que lleva el propio termo).

Saludos y gracias por tu trabajo pablo

pablozg, 2020-09-14:
Hola, por el momento tengo pensado centrarme en las pinzas, pero si en un futuro no me agradan los resultados lo tendré en cuenta.

Gracias.