pablozg/freeds #12: HomeAssistant

rdaguerre, 2020-09-01:
Hola, como estas? te felicito por este proyecto.
Tengo un mpp solar 5048gk y lo conecte por MQTT (usando ned-kelly/docker-voltronic-homeassistant) y después de ardua lucha pude hacer que mandara al servidor. Después eso si fácilmente me quedo conectado con HomeAssistant.
Debido a que tengo todos los datos ahí tanto ahora lo del inversor como el consumo a la red (por un esphome sonoff pow R1 tambien en HA)
Estaria bueno poder sacar los datos de consumo de la carga y consumo hacia la red desde un servidor mqtt …
Si tenes alguna idea de como hacerlo te agradezco
Saludos
muy amable

Ronnie

pablozg, 2020-09-02:
Hola, no uso HomeAssitant, por lo que no te puedo ayudar, pero si puedo añadir a los parámetros que publica mediante mqtt, el consumo de la carga (debes saber que es estimado ya que freeds no soporta la medición de la carga mediante pinza), ya que el consumo de red ya lo indica en el valor "wgrid" donde negativo es que está consumiendo y positivo es que está vertiendo, igual que la temperatura del inversor que es publicada en el valor "invtemp".

Siento no poder ayudarte más.

rdaguerre, 2020-09-10:
Capaz que no me explique bien. En vez de que el consumo venga por el RS485 o el inversor o etc... que el freeds lo saque de una publicacion en un servidor mqtt. Esto permitiria que todos que no sea compatible lo pueda leer de ahi y ahi si que trabaje normalmente. No se si me explique? saludos y muy amable, saludos

pablozg, 2020-09-14:
Hola eso ya lo hace usando la opción tasmota mqtt, los datos los extrae de una publicación json como la siguiente, siempre la puedes replicar con nodered o similar y elegir esa opción.

Meter:

````
Casa/PB/sdm230/tele/SENSOR = {"Time":"2020-09-14T14:54:56","ENERGY":{"TotalStartTime":"2020-05-27T21:00:58","Total":2183.985,"TotalTariff":[986.462,1197.499],"Yesterday":8.521,"Today":3.552,"ExportActive":1470.398,"ExportTariff":[512.002,958.178],"Period":1.96,"Power":2172.84,"ApparentPower":2178.72,"ReactivePower":-51.48,"Factor":1.00,"Frequency":50,"Voltage":225.6,"Current":9.68,"ImportActive":1634.135,"ImportReactive":577.035,"ExportReactive":46.551,"PhaseAngle":358.63}}
````

En el caso del meter solo necesitas crear el valor 'power' dentro de ENERGY.

Solar:
````
Casa/PB/solaxX1/tele/SENSOR = {"Time":"2020-09-14T14:58:17","ENERGY":{"TotalStartTime":"2020-09-14T14:56:07","Total":5855.4,"TotalTariff":[14638.5,20284.6],"Yesterday":17.4,"Today":6.2,"Period":0.00,"Power":2591.00,"ApparentPower":2631.12,"ReactivePower":457.70,"Factor":0.98,"Frequency":50.00,"Voltage":230.8,"Current":11.40,"SolarPower":2712.50,"Pv1Voltage":176.4,"Pv1Current":8.00,"Pv1Power":1411.20,"Pv2Voltage":185.9,"Pv2Current":7.00,"Pv2Power":1301.30,"Temperature":50.0,"Runtime":5631,"Status":"Working","Error":0},"TempUnit":"C"}
````

closed 2021-02-14 by pablozg