pablozg/freeds #19: Wibeee se reinicia

robertofilipe, 2020-12-06:
Hola Pablo, tengo ESP32 funcionando correctamente con Shelly EM pero con Wibeee se reinicia constantemente. Cuál podría ser la solución?
Gracias

pablozg, 2020-12-08:
Hola, es un fallo conocido de wibeee, pero con los últimos cambios que he realizado parece que están solucionados, lo podrás probar en la próxima versión.

robertofilipe, 2020-12-08:
Hola Pablo, muchas gracias por tu dedicación al proyecto. Tan pronto como la próxima versión esté disponible, haré la actualización. Gracias

pablozg, 2020-12-09:
Si te unes al canal de FreeDs en telegram te puedo mandar una versión de prueba para que hagas pruebas y me dices si funciona mejor en tu caso.

http://t.me/freeds_es

robertofilipe, 2020-12-09:
Hola, sí ya era parte de Telegram. Que sería genial. Enviaré un mensaje en Telegram. Gracias

closed 2021-02-14 by pablozg