pablozg/freeds #20: Enviar datos a influxdb

s118, 2021-02-13:
Hola. Antes de nada muchisimas gracias por tu aportación. Estoy montando el derivador de excedentes, y viendo el estado tan desesperante de solaxcloud y de su app para monitorizar la producción solar y el consumo/inyección de la casa, he pensado si sería posible enviar desde el derivador datos de producción solar y de consumo/inyección a una base de datos influxdb que tengo montada en una raspberry almacenando datos de varios sensores, para despues representarlos en grafana.
La raspberry está conectada al router por LAN, aunque también puedo conectarla a la wifi del pocket solax.
Muchas gracias.

tfael, 2021-02-14:
Sí, no estaría mal una integración con Homeassistant. De allí ya se pasaría a InfluxDB/Grafana.

pablozg, 2021-02-14:
Hola, en estos momentos estoy liado con temas de trabajo y estoy ultimando la última versión que voy a publicar antes de centrarme en el trabajo y serán unos dos o tres meses en los que sólo tocaré el proyecto para temas urgentes.

A corto plazo no creo que pueda implementar nada directamente, pero si que puedes utilizar los datos que publico por mqtt para usando node-red por ejemplo enviarlos a influxdb.