pablozg/freeds #21: Homeassistant

tfael, 2021-02-14:
Hola, algún plan de desarrollar una integración en Homeassistant?

pablozg, 2021-02-14:
Hola, en estos momentos estoy liado con temas de trabajo y estoy ultimando la última versión que voy a publicar antes de centrarme en el trabajo y serán unos dos o tres meses en los que sólo tocaré el proyecto para temas urgentes.

Respecto a HA no creo que vaya a desarrollar ninguna implementación a no ser que alguien lo haga por su cuenta, ya que uso Domoticz en su lugar y como te digo ahora mismo no tengo demasiado tiempo libre, pero te animo a que lo intentes tú directamente.

Un saludo.

jorgestima, 2021-09-28:
I'm also using HA and indeed, it would be great to have some integration. For example, if just two things were possible to integrate in HA as ON/OFF button and target power value, that would be really good.

fizcris, 2021-09-28:
Hi, 

I have started to do the integration here [aquí](https://github.com/fizcris/home-assistant-freeds).

But since then I have found [MQTT discovery](https://www.home-assistant.io/docs/mqtt/discovery/), whit this there is no need anymore for a has integration as if MQTT discovery is included in the ESP32 all the sensors would be automatically available in hassio as long as they are using the same broker.

pablozg, 2021-09-28:
> I'm also using HA and indeed, it would be great to have some integration. For example, if just two things were possible to integrate in HA as ON/OFF button and target power value, that would be really good.

Hi, I suggest you to suscribe to the [Telegram Channel](https://t.me/freeds_es) and ask for the HA integration, I think someone it has done.

You can use a 'POST' message to freeDS to switch ON/OFF with http://FREEDSIP/tooglebuttons?data=6
And switch it between AUTO / MAN with http://FREEDSIP/tooglebuttons?data=7

Or use mqtt as fizcris said, in the future we can integrate MQTT discovery topics.

jorgestima, 2021-09-28:
I know that MQTT is a possibility and very powerfull but more complicated to configure. As only recently I've entered in HA world, it is not easy for me to configure that. So, if a simple tool in HACS would be available, for me it would be great.

jorgestima, 2021-09-28:
Many thanks @fizcris 

I will have a look on that. I already have all that data because I have my Goodwe solar inverter communicating using GoodWe Inverter Solar Sensor UDP installed by HACS.

What I was really looking for was to turn ON/OFF the excess power compensation and adjust the target power value using HA. Therefore, I could program and interact with all my other devices in just one single monitoring/control platform.

Do you have any idea on how this could be made in HA?

fizcris, 2021-09-29:
Hi,

That's not possible to do at the moment for any third party app regarding
an Goodwe inverter as we are still understanding how to change the
registers of the inverter using inverse engineering.

I think it will be available shortly as most of the protocol is understood
now but I suggest you to join the discord group of the Goodwe integration
to follow/assist with the development of such avanced features.

Regards

jorgestima, 2021-09-29:
Well, I think that I didn't explain correctly. The idea is not to change or interact with inverter. What I was saying is putting HA sending commands to MCU/FreeDS to turn ON/OFF the excess power compensation and adjust the target power value.

So, you already sent the code to read values from MCU using MQTT in HA. What would be the code to send command to turn ON/OFF compensation and change target power value?

IvanSanchez, 2023-04-07:
Hi, folks,

I've found out that setting up MQTT is a bit too fiddly, so in the end I've created a HASS integration that relies on the `http://$ip/events` endpoint - values available at that endpoint are automatically exposed as HASS sensors.

It may be buggy at the moment, and can only handle one FreeDS in the same network (because of #82), but the basics work.

See https://gitlab.com/IvanSanchez/homeassistant-freeds