pablozg/freeds #24: Issues with Wibeee

quissanga, 2021-04-18:
Hi,

I found out Wibeee server crashes after a while. The only way to fix is to hard reset Wibeee to be able to poll the data from http://<ip-address>/en/status.xml.

I don't know if it has anything to do with the way the HTTP GET request is done via the FreeDS.
At first I thought it was because of agressive default polling (1500ms) but i see it's something else.
When I do it non stop from a PC using Chrome it does not crash wibeee http server but FreeDS after a while manages to crash it (the wibeee server)...

  Does Wibeee Modbus work ok? It's still in desarollo :)
  
  Muchas gracias por tu trabajo!

quissanga, 2021-04-18:
Fue un bug en la versión de firmware de Wibeee fw: 4.4.150
Actualicé Wibeee a la versión 4.4.155.

problema resuelto.

Gracias

pablozg, 2021-04-23:
Efectivamente, el servidor http de wibeee da muchos problemas según la versión, me alegro que esté solucionado.

closed 2021-04-23 by pablozg