pablozg/freeds #25: Compilación

Susejer, 2021-04-21:
Hola, 

¿Está el código completo para ser compilado? No he utilizado mucho el IDE de arduino y me saltan errores. El último está relacionado con la función INFOV, la he buscado con el buscador de GitHub y no aparece. ¿Dónde se define esa función?¿Están todos los includes? Gracias.

pablozg, 2021-04-23:
El código está completo, pero para compilarlo necesitas usar visual studio code + platformio, cuando lo tengas todo instalado abres la carpeta con el código y debe compilar sin problemas.

La última versión se encuentra en la rama PID.

Susejer, 2021-04-25:
.Gracias. 
Pues la verdad es que ha sido una confusión por mi parte al ver los ficheros .ino he pensado que lo habías realizado en el IDE de arduino. La verdad es que no tengo expericencia copiando proyectos de otros usuarios, es la primera vez y mira que tengo muchos proyectos con PlatformIO, pero la extensión me ha dejado "kao".

Copiando y pegando no funciona. He tenido que hacerlo con GitHub, añadir el proyecto y luego incluir ArduinoJson y EmonLib. Ha compilado con algunas advertencias pero funciona. Voy a ver si se puede hacer la modificación de forma "limpia". Ya te aviso si hay éxito.

¿Tiene alguna ventaja el uso de ficheros ino con PlatformIO?

pablozg, 2021-04-25:
Como he indicado en el grupo en estas fechas estoy bastante ocupado en el trabajo y no tengo tiempo para el proyecto (aunque es solo de forma temporal), la versión PID que está subida no es la más actualizada ya que tengo algunos cambios hechos que no he subido todavía.

El usar .ino fue porque empecé a programar en el IDE de arduino y se quedó un poco corto en poco tiempo por eso pasé a platformio, pero no he tenido tiempo de pasar la estructura del proyecto a puramente c++, por eso siguen estando los fuentes con la extensión .ino, por lo que en un futuro más bien lejano trataré de pasarlo a c++ o si te animas lo puedes hacer tú.

¿Qué quieres modificar exactamente?

Susejer, 2021-05-05:
Pues un compañero del grupo de telegram me ha pedido una modificación para hacer en realmente se ha un gestor de excedentes universal. Yo no utilizo FreeDS, ya que tengo mi propio programa de control basado en el balance neto.

Básicamente he cambiado unas cuantas líneas para cargar los contadores del inversor que me han pedido y hacer balance horario para no pagar los picos.  Si dejas la última versión aquí en GitHub la utilizaría.

El pasar de uno a c++ es una tarea que llevaría tiempo y conocimiento del código y yo también ando justo de tiempo.

Muchas gracias!!

closed 2021-08-23 by pablozg