pablozg/freeds #28: Errores en la compilación y en la ejecución

fjsalc, 2021-05-20:
Hola, estoy tratando de hacer algunas pequeñas modificaciones, pero la verdad, es que me está resultando imposible porque de partida no va bien, lo que me hace sospechar que las librerías que se utilizan, o no están actualizadas, o bien no son con las que realmente se compilan finalmente con el programa. Voy a explicar paso a paso los errores con los que me he ido encontrando.

1. Me he bajado la versión PID como aconseja @pablozg, y sin hacer ningún cambio intento compilarla en **platformio**. Primer error: no compila porque dice que no encuentra el archivo _**preprocessor.hpp**_ de la librería **ArduinoJson**. Lo busco, y efectivamente, no está. Para resolverlo actualizo la librería a la última versión, que sí trae dicho archivo. Problema resuelto, ahora ya compila por completo **FreeDS**.
2. Cuando paso el programa al **wifi kit 32** y empieza a ejecutarse, por el terminal aparece un error: "_**Error al analizar el archivo de lenguajes**_". No parece un error importante porque permite que se siga ejecutando el programa.
3. Lo peor viene ahora. Consigo conectarme vía wifi al AP creado FreeDS, aunque en el display no aparece el texto "**_Conectarse al   SSID_**" ni la IP a la que conectarse. De todos modos me puedo conectar a la web en la IP 192.168.1.4, selecciono la wifi de mi casa o la del inversor, y no conecta con ninguna de las dos. Aparece el mensaje en el display de "**_WIFI ERROR_**". Y en el terminal conectado a platformio aparece ésto:
[image]

Agradecería que alguien pudiera echarme una mano en este punto en el que ya no veo una posible solución. Espero hacer una pequeña contribución al proyecto.

Gracias

pablozg, 2021-05-20:
Hola, efectivamente ArduinoJson no viene incluido en el repositorio, respecto a los textos, debes compilar el fichero spiffs.bin y subirlo también al esp32.

Primero lo compilas desde la consola de platformio con  `pio run -t buildfs`
Y luego lo subes con `pio run -t uploadfs` 

Sobre el error al conectar puede ser por señal o por la clave que tenga el ssid, pero dudo mucho que sea problema de compilación.

También decirte que la versión subida al repositorio no está totalmente actualizada, ahora mismo estoy liado con temas de trabajo, pero en un mes o así ya podré retomar de nuevo el proyecto.

¿Qué cambios quieres hacer?

Saludos.

fjsalc, 2021-05-21:
Gracias @pablozg. 

He compilado y subido el fichero **_spiffs_**.bin y ahora los mensajes aparecen correctamente en el display. Sin embargo, sigo teniendo problemas con las conexión a las wifi. He probado varias veces  y siempre da el mismo error, el que puse en el mensaje anterior. Por supuesto no es problema de la contraseña, e incluso para probar si es problema de hardware, he subido al ESP32 los binarios de la versión PID y funcionan correctamente, se conectan a las wifi. Por eso sigo pensando que es un problema de versiones de librerías wifi o algo así.

Lo que quiero hacer es que dé la posibilidad de tener en cuenta la compensación por horas. Todo el que tiene compensación de excedentes reconocido por la comercializadora y distribuidora, se le hace primero una compensación dentro de cada hora, donde los Kwh vertidos se compensan con los consumidos. Este el mejor modo de rentabilizar los KWh que produce tu instalación solar. Por ello, sería interesante que el freeds calculara el balance de Kwh vertidos y consumidos en cada hora para no activar el consumo de las salidas o del derivador si en la hora que corre se ha consumido más de lo que se ha vertido. Como esta situación viene bien sólo en determinadas circunstancias, habría que poner un interruptor en la web para decidir si se activa o no este modo.
 Yo tengo hecho ya los cambios en los fuentes para hacer la contabilidad de este balance por horas y hacer que se pare o activen las salidas y el derivador. Lo único que me queda es buscarle un sitio en el display para mostrar cuál es este balance en cada momento y el activar o desactivar este modo de funcionamiento en la web.  Si quieres te paso lo que he hecho o lo subo como un fork para que puedas utilizarlo y no te calientes mucho la cabeza, aunque no es muy difícil.

pablozg, 2021-05-23:
Hola he actualizado el código fuente a la versión que tengo actualmente y he subido un archivo .zip con las librerias tal y como las tengo yo en mi sistema.

Prueba a compilar con las librerías que he subido, si te sigue fallando ya debe ser problema de configuración del platformio.

Respecto a los cambios, cuando lo tengas todo compilado y probado, crea un pull request para que me sea más sencillo comprobar los cambios y hacer los ajustes si fueran necesarios.

Gracias.

fjsalc, 2021-05-28:
Hola de nuevo. Bien parece que ahora va todo mejor. Compila sin problemas y pasa el código al ESP32. El único inconveniente es que yo tengo un Solax (no híbrido) y cuando lo conecto en modo local lee perfectamente los datos de producción y de consumo. Sin embargo, cuando lo conecto a través del ESP01 (no local), se conecta al inversor, pero ya no le llegan bien los parámetros. Parece como si no interpretara bien el vector de datos que le llega. 

En el código fuente que subiste no está la fuente de los programas del ESP01, no? Si es así, te pediría que lo subieras también, o al menos el binario, porque para mí es más fácil hacer los cambios y correcciones sobre la conexión no local, es decir, usando el ESP01. (Por cierto, el ESP01 está bien porque funciona perfectamente con OpenDS en conexión no local)

Gracias.

pablozg, 2021-05-28:
El binario para el esp01 lo tienes en el zip de las releases: [https://github.com/pablozg/freeds/releases/download/1.0.7/FreeDS_1.0.7Rev2.zip](url)

Ten en cuenta que el binario del esp01 ya no es compatible con opends por eso no te lee los datos, en cuanto lo programes, tendrás que elegir en la configuración el ssid del solax para que lo envíe al esp01.

Si los datos no te salen de forma correcta prueba a enviar el comando "solaxVersion 2" desde la consola o "solaxVersion 3", el comando depende de la versión de solax que tengas.

fjsalc, 2021-05-29:
Gracias  @pablozg. 

El comando era lo que faltaba. Ya puedo compilar y comprobar los cambios. Te aviso cuando los tenga subidas las modificaciones.

closed 2021-08-23 by pablozg