pablozg/freeds #29: Error conectar wifi solax v2 local

polcino, 2021-06-07:
Hola, buenas noches. Ante todo dar las gracias a pablozg y a la gente que esta trabajando en este proyecto.

No se si estoy haciendo algo mal, pero despues de seleccionar la wifi local del solax y meter la clave, el ESP32 reinicia y siempre me dices ERROR WIFI.

He probado con la 1.0.7 beta, la 1.0.7 Final y la 1.0.6 todas con el mismo resultado.

He cambiado la clave de la wifi local, mismo error, sin embargo a la wifi de casa si conecta bien.

No se si hago algo mal o es algun otro tipo de problema.

Saludos

xelect75, 2021-06-08:
Olá, parabens por todo o trabalho neste projecto.
Eu não consigo conectar na minha rede DHCP, dá sempre "Error Wifi"
quais são os requisitos para conectar correctamente numa rede wifi?
Obrigado,
Nelson

closed 2021-11-18 by pablozg

RoyGeova07, 2024-08-16:
Hola, todo esta bien