pablozg/freeds #30: Freeds con Baterias

jeamobarb, 2021-06-10:
En mi instalación solar, tengo placas, batería, batrium, victron, fronius y FREEDS.
FREEDS, lo tengo configurado para que empiece a derivar cuando la batería llegue al 80% de la batería.
El problema está hasta que no llega a ese 80%. Las baterías, no usan el 100% de la energía solar, cuando las baterías están bajas, no hay problema ya que utiliza mucha potencia de las placas, pero a medida que se van cargando, esa potencia, se va reduciendo hasta el punto que quizás solo consumen de las placas 300W cuando las placas podrían dar 10 veces mas, pero como las baterías no han llegado al 80%, FEEDS no empieza a derivar con la consiguiente perdida de energía que se podría aprovechar para derivarla a FREEDS.
He visto que en la configuración de FREEDS, en el mismo apartado donde se le indica que empiece a derivar a partir de un porcentaje de las baterías, hay otra opción que se dice "Consumo máximo de batería" donde parece ser que se le indican los watios máximos, creo que se refiere al consumo de energía máxima que se utilizará para cargar las baterías, aunque no estoy seguro, el caso es q esa opción la tengo deshabilitada.
Hay alguna manera para mejorar mi sistema que desconozca y poder aprovechar mas la energía solar?
P.D.: perdón si no es el sitio donde preguntar

jeamobarb, 2021-06-10:
Solucionado por Telegram, gracias

closed 2021-08-23 by pablozg

ruisilva93, 2025-04-06:
Como? Parece o mesmo problema que tenho.