pablozg/freeds #35: Producción solar de valores negativos a positivos

red-machine95, 2021-08-10:
Hola, ¡felicidades por el fantástico proyecto!
Me gustaría dejar la sugerencia, de implementar la conversión de valores negativos de producción solar, por valores positivos.

Sé que existe la posibilidad de ejecutar la pinza de lectura, pero si hago este cambio, ya no tengo la posibilidad de ver la inyección en la red eléctrica, a través de la aplicación de Shelly.

Por ello, dejo la sugerencia de que en FreeDS existe esta posibilidad, como ya ocurre en el valor de la red (positivo o negativo).

(Perdón por mi español, intenté expresarme de la mejor manera 😅)

pablozg, 2021-08-23:
Hola, los valores de energía solar solo son a modo informativo y solo admiten valores positivos, ya que es la energía que produce el inversor o la suma de los strings.

Por lo que no entiendo tu necesidad de ver los valores en negativo.

Te recomiendo unirte al grupo de Telegram, el enlace está en la página web del proyecto y ahí puedes realizar también sugerencias o preguntar dudas.

Un saludo.

red-machine95, 2021-08-23:
¡Hola! Es como dije, para mí me da una forma de ver los valores negativos porque en la aplicación shelly puedo ver el excedente solar. ¡Con valores positivos no es posible ver esta información!

Por lo tanto, dé la sugerencia de potencia en FreeDS, convierta los valores negativos en positivos 😄

Pero no se preocupe, para mí es solo informativo, lo que importa es que FreeDS haga su trabajo y tenga un excedente solar muy bajo 😊

closed 2021-11-18 by pablozg