pablozg/freeds #36: Backup configuración da error 404

ayasystems, 2021-08-23:
Buenas,

Estaba probando la beta 1.0.7 Beta Rev. D

Al ir a exportar la configuración en la página de actualización, da error 404

[image]

Estaba mirando si hay un handler para downloadBackup y si lo hay 

En la consola serial da el siguiente error

```
I0.00,O0.00,T0.00,PWM0,P0,MODE:0,DIRECTION:1
[E][vfs_api.cpp:265] VFSFileImpl(): fopen(/spiffs/config_freeds_a4f4_1.0.7 Beta Rev. D.bin) failed
Write 0 bytes
[E][vfs_api.cpp:135] remove(): /config_freeds_a4f4_1.0.7 Beta Rev. D.bin does not exists or is directory
I0.00,O0.00,T0.00,PWM0,P0,MODE:0,DIRECTION:1
```

Un saludo

pablozg, 2021-08-23:
Hola, esa versión era un beta, la última disponible es esta:

https://github.com/pablozg/freeds/releases/download/1.0.7/FreeDS_1.0.7Rev2.zip

ayasystems, 2021-08-24:
Sorry, no vi que más abajo en las release había otra más moderna ;)

Ya probé la que me indicas y todo OK!

closed 2021-08-24 by ayasystems