pablozg/freeds #38: Huawei Smartlogger

fapgomes, 2021-09-19:
Ola,

Puedes adicionar el suporte a huawei smartlogger 3000 ? És identico a

```
registerData huaweiRegisters[] = {
    &inverter.wgrid, 0, 37113, 2, S32FIX0,
    &inverter.wtoday, 0, 32114, 2, U32FIX2,
    &inverter.wsolar, 0, 32064, 2, S32FIX0,
    &inverter.pv1v, 0, 32016, 1, S16FIX1,
    &inverter.pv1c, 0, 32017, 1, S16FIX2,
    &inverter.pv2v, 0, 32018, 1, S16FIX1,
    &inverter.pv2c, 0, 32019, 1, S16FIX2,
    &inverter.temperature, 0, 32087, 1, S16FIX1,
    &inverter.batteryWatts, 0, 37001, 2, S32FIX0
};
```

Pero lo que pude apurar és necessário passar el id del dispositivo (pues puede ser el powermeter, o el inversor, o mismo el smartlogger. Será algo deste estilo:

```
registerData huaweiRegisters[] = {
    &inverter.wgrid, 0, 32278, 2, S32FIX0,    # valor proveniente del power meter
    &inverter.wtoday, 0, 40562, 2, U32FIX2,  # valor proveniente del smartlogger
    &inverter.wsolar, 0, 40525, 2, S32FIX0,   # valor proveniente del smartlogger
};
```

Puede testar, se lo quieres :)

Buen trabajo,
Fernando Gomes

pablozg, 2021-09-20:
Para implementarlo necesito saber exactamente el id la dirección del registro y la variable a la que corresponde, cuando tengas toda esa información dímelo y lo podré integrar, pero asegúrate que es compatible con todos los smartlogger 3000 y no requiere de una instalación o configuración especifica.

gracias.

fapgomes, 2021-09-20:
Depende, el id es configurable, en mi caso es 0 para el smartlogger y 11 para el powermeter, pero debe ser un campo configurable, para poder atender a todos:

[image]

Muchas Gracias,
Fernando Gomes

fapgomes, 2021-09-20:
Ai aparece el id del inverter, pero para freeds, pienso que no necessitamos de nada dele.

fapgomes, 2021-09-25:
@pablozg Lo ideal era, para modbus tcp, para cada ítem que sea necesario, poder definir el id y el canal y tener modelos ya fijos y conocidos, pero no limitados a estos.

closed 2022-09-14 by pablozg