pablozg/freeds #39: FreeDS only for monitoring

jorgestima, 2021-09-28:
First, I'm very impressed with FreeDS than I'm using. Congratulations for the excellent work on this!!!

I was wonder if it would be possible to develop a simple version only for monitoring purposes, that is, using the ESP32 with OLED display just to display data from the inverter as solar power, grid power, house consumption, battery power, battery SOC...

The idea is to use the entire area of this OLED display just to show these parameters (maybe we could select which variables we want to see in the OLED display).

pablozg, 2021-09-28:
I will think about it, but isn't a priority at this moment, however you can try to change the code and do it yourself.

Thanks.

fizcris, 2021-09-28:
I think this would overcomplicate things in an embedded device. The purpose of freeds is ideal for what an MCU has been conceived.  You can use freeds with home assistant for the purposes that you are stating.

jorgestima, 2021-09-28:
I fully understand that it is not a priority. I don't have enough knowledge to change the code by myself but, I suppose that as the MCU is only used to display data (already shows solar and grid power), this could be something relatively simple when compared to all features that FreeDS has now.

I do have HA but for the purpose of continuous monitoring, it is much simpler having this very small and low power MCU+OLED display showing data in a small corner of my kitchen than, in comparison of having a much bigger, bulkier and expensive smartphone/tablet device always powered on...

closed 2021-11-18 by pablozg