pablozg/freeds #46: Modbus TCP Custom

hugomon, 2021-11-10:
### ANTECEDENTES
Una de las formas más comunes de conectar el freeds con el inversor es a través del protocolo Modbus TCP (o del RTU y usando un conversor RTU-TCP).
Por ejemplo los valores que el freeds toma del inversor pueden ser:
_inverter.acIn, inverter.acOut, inverter.batterySoC, inverter.batteryWatts, inverter.loadWatts, inverter.pv1c, inverter.pv1v, inverter.pv2c, inverter.pv2v, inverter.pw1, inverter.pw2, inverter.temperature, inverter.wgrid, inverter.wsolar, inverter.wtoday_, entre otros.
Pero bastaría como mínimo con _inverter.wgrid_ para poder funcionar.

### PROBLEMA
Existen multitud de inversores (aún) no soportados, dado que en cada uno varían las direcciones de los registros así como el tipo de datos.

### PROPUESTA
Como mejora propondría un campo de elección de adquisición de datos **Modbus TCP Custom,** con varios campos donde seleccionar la dirección, tipo de datos, etc. de algunos de los valores más útiles a extraer del inversor (mínimo _inverter.wgrid_, quizás    _inverter.wsolar_).
Así cada uno podría configurar los valores de su inversor en particular sin tener que cambiar el código para cada uno nuevo que salga, y se puede generar una tabla de consulta donde los que lo hayan hecho aporten su configuración.

Linux-RISC, 2022-10-13:
Incluso podría tenerse la posibilidad de importar un XML con la descripción de los registros, de tal manera que la comunidad pueda ir contribuyendo con diferentes inversores, medidores de energía, ...

Algo así:

```
<holding_registers>
    <holding_register>
        <register>5000</register>
        <decimals>2</decimals>
        <units>V</units>
    </holding_register>
    <holding_register>
        <register>5001</register>
        <decimals>1</decimals>
        <units>A</units>
    </holding_register>
</holding_registers>
```

Si por ejemplo el registro 5000 tiene el valor 22012, el valor real sería 220.12 V

Linux-RISC, 2022-10-13:
También propongo que el FreeDS, además de ser servidor MQTT, sea servidor MODBUS/TCP o MODBUS RTU sobre TCP.

Este protocolo es muy utilizado en automatización para enviar datos entre PLCs, podría servir para que leyese los datos Home Assistant o un PLC, por ejemplo.

La descripción de los datos exportados podría ser en el formato especificado anteriormente.

pablozg, 2022-10-13:
Aunque lo que sugieres tiene sentido, por el momento no es viable al requerir un gran cambio en el código, además como único desarrollador no puedo dedicarle tiempo a esto, quizá en un futuro se haga, pero el poco tiempo que tengo para dedicarle al proyecto debe ser en incorporar y mejorar otras funcionalidades que tengo en el tintero desde hace tiempo.

Pero como te digo lo tengo apuntado ya como mejoras para un futuro.

Gracias por tus sugerencias.