pablozg/freeds #51: Configurar data e hora

jescampos, 2022-02-08:
É possível configurar data e hora? Definir um time zone? Ntp server? 

É possível aceder por SSH para configurar directamente no S.O.?

miselfi1, 2022-02-10:
e ler o manual?

jescampos, 2022-02-10:
Não fala nada sobre ntp e SSH.

miselfi1, 2022-02-10:
é feito na cansola web
vê o codigo fonte tem lá um ficheiro que se chama timeQUAQUERCOIAS.txt,
copia a linha que quiseres e cola na consola...
o ntp é feito por essa linha nativamente

jescampos, 2022-02-10:
Afinal estava no Manual. Obrigado mais uma vez. 👍

closed 2022-02-10 by jescampos

miselfi1, 2022-02-10:
👍

miselfi1, 2026-01-16:
> I tried sending the text from the TimeZones.txt, I mean the line I
> interest, into the console... but nothing is happening:
>
> "Europe/Bucharest","EET-2EEST,M3.5.0/3,M10.5.0/4"
>
> I sent this in console, and nothing happening...

use only

EET-2EEST,M3.5.0/3,M10.5.0/4

valimaties, 2026-01-16:
> hi
> 
> using wrong code...
> 
> "Asia/Nicosia","EET-2EEST,M3.5.0/3,M10.5.0/4"
> 
> that is for asia/nicosia...
> 
> Right one for europe/bucarest is:
> 
> "Europe/Budapest","CET-1CEST,M3.5.0,M10.5.0/3"
> […](#)

[image]

This is from TimeZones.txt, in the zip file of Pablo's firmware upgrade file:

[image]

Bucharest is the same as Athens.

And after using that code, it displays correct time and date.

miselfi1, 2026-01-16:
somethings not right...
[image: Screenshot 2026-01-16 at 11.27.03.png]
[image: Screenshot 2026-01-16 at 11.27.42.png]