pablozg/freeds #53: Battery SOC with Goodwe ES/EM

jorgestima, 2022-02-23:
In my case that I have a inverter Goodwe EM, the battery SOC value in FreeDS is wrong. For example, it shows 13568% when the real value is 53%. It seems that the shown value must be divided by 256 to get the corret SOC value.

Also, it shows the battery with negative power and providing power, which is wrong, because battery negative power means that it is charging.

[image]

pablozg, 2022-09-14:
Please, check with the last beta again.

ibm8999, 2022-10-04:
Buenas noches Pablo
Lo comenté por el grupo y te lo pongo por aquí también. 
La beta 12 funcionando perfectamente en GoodWe. 

Muchas gracias por tu trabajo 
Un saludo
Paco

closed 2022-10-12 by pablozg