pablozg/freeds #55: Huawei RTU

epserra, 2022-04-10:
Gracias por el entorno.
Puedes para los que hemos perdido el modbus tcp de Huawei incluir la conexión de modbus-rtu a traves del conversor modbus ? 
Gracias

diyeasy-es, 2022-08-03:
Hola, acabo de hacerle una modificación partiendo del branch PID que hay en el repositorio y parece que me está funcionando.

En mi caso tomé un ESP32 y le soldé el conversor modbus que recomiendan en la WEB siguiendo el diagrama de conexión de la web  http://freeds.es/img/WIFI%20Kit%2032.jpg

Despues conecté el pin 1 del conector del inversor (B) con el pin (B) del conversor modbus y el pin 2 del conector del inversor (A) con el pin (A) del conversor modbus. Si esos pines los hubiera usado tu instalador creo que podrias probar con el pin 3 y 4 y si los tuvieras libres

Adjunto los binarios por si los quieres probar y te funcionan... Supongo que tendrás que flashearlos por USB como indican en la web

[FreeDS_1.1.0010_ModHuaweiRTU.zip](https://github.com/pablozg/freeds/files/9252240/FreeDS_1.1.0010_ModHuaweiRTU.zip)

Muchas gracias a los desarrolladores por compartir las fuentes que me han permitido hacer la modificación. Copio y pego la función que he creado por si quieren incluirla en el release oficial

const uint16_t HuaweiRTU_start_addresses[] {
  37113,   // HUAWEI Wgrid (2)             [W]    0
  32114,   // HUAWEI Wtoday(2)             [W]    1
  32064,   // HUAWEI Wsolar(2)             [W]   2
  32016,   // HUAWEI PV1v (1)      [KVAR] 3
  32017,   // HUAWEI PV1c (1)               4
  32018,   // HUAWEI PV2v (1)           [Hz]   5
  32019,   // HUAWEI PV2c (1)       [kWh]  6
  32087,   // HUAWEI Temperature (1)       [kWh]  7
  37001,   // HUAWEI BatteryW (2)    
};

void huaweiRTU(void)
{
  bool data_ready = modbusReceiveReady();
    
  if (data_ready) {
    uint8_t buffer[14];  // At least 5 + (2 * 2) = 9

    uint32_t error = modbusReceiveBuffer(buffer, 2); 
 //   for(int i = 0; i<8;i++){Serial.print(buffer[i],HEX);Serial.print(" ");}
 //   Serial.println();

    if (error) {
      INFOV("HUAWEI RTU error: %i\n", error);
    } else {
      int32_t value32 = int32_t(
            (uint8_t)(buffer[3]) << 24 |
            (uint8_t)(buffer[4]) << 16 |
            (uint8_t)(buffer[5]) << 8 |
            (uint8_t)(buffer[6]));

      int16_t value16 = int16_t(
            (uint8_t)(buffer[3]) << 8 |
            (uint8_t)(buffer[4]));

      switch(meter.read_state) {
        case 0: //power_meter_active_power
          inverter.wgrid = (float)(value32 / 1.0); //Gain 1
          //Serial.print("power_meter_active_power: "); Serial.println(inverter.wgrid, 0);
          break;

        case 1: //daily_yield_energy
          inverter.wtoday = (float)(value32 / 100.0); //Gain 100
          //Serial.print("daily_yield_energy: "); Serial.println(inverter.wtoday,2);
          break;

        case 2: //input_power
          inverter.wsolar = (float)(value32 / 1.0); //Gain 1
          //Serial.print("input_power: "); Serial.println(inverter.wsolar,0);
          break;

        case 3://pv_01_voltage
          inverter.pv1v = (float)(value16 / 10.0); //Gain 10
          //Serial.print("pv_01_voltage: "); Serial.println(inverter.pv1v,1);
          break;

        case 4://pv_01_current
          inverter.pv1c = (float)(value16 / 100.0); //Gain 100
          //Serial.print("pv_01_current: "); Serial.println(inverter.pv1c,2);
          break;

        case 5://pv_02_voltage
          inverter.pv2v = (float)(value16 / 10.0); //Gain 10
          //Serial.print("pv_02_voltage: "); Serial.println(inverter.pv2v,1);
          break;

        case 6://pv_02_current
          inverter.pv2c = (float)(value16 / 100.0); //Gain 100
          //Serial.print("pv_02_current: "); Serial.println(inverter.pv2c,2);
          break;

        case 7: //internal_temperature
          inverter.temperature = (float)(value16 / 10.0); //Gain 10
          //Serial.print("internal_temperature: "); Serial.println(inverter.temperature,1);
          break;

        case 8: //storage_unit_1_charge_discharge_power
          inverter.batteryWatts = (float)(value32 / 1.0); //Gain 1
          //Serial.print("storage_unit_1_charge_discharge_power: "); Serial.println(inverter.batteryWatts,0);
          break;
      }

      meter.read_state++;

      // Error.VariacionDatos = false;
      Error.RecepcionDatos = false;
      timers.ErrorRecepcionDatos = millis();
      
      if (meter.read_state == 9) { // 8
        meter.read_state = 0;

        inverter.pw1 = inverter.pv1v * inverter.pv1c;
        inverter.pw2 = inverter.pv2v * inverter.pv2c;
      }
    }
  } // end data ready

  if (0 == meter.send_retry || data_ready) {
    meter.send_retry = 5;
    modbusSend(config.idMeter, 0x03, HuaweiRTU_start_addresses[meter.read_state], 2);
  } else {
    meter.send_retry--;
  }
}

pablozg, 2022-08-04:
@diyeasy-es El código que se encuentra en el repositorio no es el correspondiente a la Beta 10, contactame por telegram y te pasaré una versión de la última beta con la modificación que has hecho para que compruebes si funciona correctamente, así podré liberar una nueva versión con ese cambio.

crywolf87, 2022-10-03:
@pablozg Me ofrezco a probar el software modbus huawei y deja feedback de la operación si me permites probarlo, gracias

pablozg, 2022-10-04:
> @pablozg Me ofrezco a probar el software modbus huawei y deja feedback de la operación si me permites probarlo, gracias

Sólo tienes que bajarte la última beta disponible para probarlo.

Si no eres miembro del grupo de telegram puedes unirte mediante este enlace: https://t.me/freeds_es

geo99918, 2024-04-29:
Hi guys, 
I'm also interested in the Modbus RTU version for Huawei, reason being that the Modbus TCP is not 100% stable.
Where do we get the Modbus RTU version to test it ?
Thank you.