pablozg/freeds #56: Heltec WiFi Kit 32 reboot loop after soldering pin headers

luisbarreto, 2022-06-03:
Hi Everyone,
First of all thank you for all the work around this excellent open source project!

I've been running the Heltec WiFi Kit 32 (v2) since the end of year 2021 just connected to an USB power source and my home wifi and internal MQTT.

Recently I started to have a little spare time to build my box.
After soldering the WiFi Kit 32 the pin headers, the connection to (previously configured) wifi keeps rebooting eventually connecting to the configured wifi about 5% of the time (at most).

When this occurred to the board I was using for several months without issues I thought the I could have bricked the board due to some short-circuit, excessive heat or something like that. 
Since then I flashed the firmware (1.0.7 Final Rev 2) without pins soldered and after confirming everything was working I then soldered the pin headers to 4 additional wifi kits and all experience the same issue. 

I consider myself pretty confident at soldering, used lower than I usually use soldering temperature (320ºC), as little solder as possible, nor short circuits and in the last boar I only soldered the pins to power, PWM, DS18B20 and external temporary switch.

If I let the boards turned on, they eventually connect to my wifi, but most of the time they just reboot time after time and are not connected.

at the console I get:

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10044
load:0x40080400,len:5900
entry 0x400806ac
Testing EEPROM Library
EEPROM Size: 2048 bytes
Current language: es
Connecting to Wi-Fi...
[I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: homeNetwork
[I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: MIWIFI2
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
[WiFi-event] event: 0
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
[WiFi-event] event: 2
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 1 - SCAN_DONE
[WiFi-event] event: 1
[I][WiFiMulti.cpp:114] run(): [WIFI] scan done
[I][WiFiMulti.cpp:119] run(): [WIFI] 9 networks found
[D][WiFiMulti.cpp:151] run():        0: [11][xx:xx:xx:xx:xx:xx] homeNetwork_2 (-45) *
[D][WiFiMulti.cpp:149] run():  --->   1: [11][xx:xx:xx:xx:xx:xx] homeNetwork (-46) *
[D][WiFiMulti.cpp:151] run():        2: [11][xx:xx:xx:xx:xx:xx] REDACTED (-82) *
[D][WiFiMulti.cpp:149] run():  --->   3: [1][xx:xx:xx:xx:xx:xx] homeNetwork (-87) *
[D][WiFiMulti.cpp:151] run():        4: [1][xx:xx:xx:xx:xx:xx] REDACTED (-88) *
[D][WiFiMulti.cpp:151] run():        5: [1][xx:xx:xx:xx:xx:xx] REDACTED (-89)  
[D][WiFiMulti.cpp:151] run():        6: [1][xx:xx:xx:xx:xx:xx] REDACTED (-90)  
[D][WiFiMulti.cpp:151] run():        7: [11][xx:xx:xx:xx:xx:xx] REDACTED (-92) *
[D][WiFiMulti.cpp:151] run():        8: [11][xx:xx:xx:xx:xx:xx] REDACTED (-93) *
[I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: xx:xx:xx:xx:xx:xx SSID: homeNetwork Channel: 11 (-46)
E (11625) wifi:Set status to INIT
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:391] _eventCallback(): Reason: 203 - ASSOC_FAIL
[WiFi-event] event: 5
WiFi lost connection
[E][WiFiMulti.cpp:184] run(): [WIFI] Connecting Failed.
.[D][WiFiMulti.cpp:195] run(): [WIFI] delete old wifi config...
[D][WiFiMulti.cpp:198] run(): [WIFI] start scan
...[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 1 - SCAN_DONE
[WiFi-event] event: 1
.[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 1 - SCAN_DONE
[WiFi-event] event: 1
[I][WiFiMulti.cpp:114] run(): [WIFI] scan done
[I][WiFiMulti.cpp:119] run(): [WIFI] 11 networks found
[D][WiFiMulti.cpp:151] run():        0: [11][xx:xx:xx:xx:xx:xx] homeNetwork_2 (-47) *
[D][WiFiMulti.cpp:149] run():  --->   1: [11][xx:xx:xx:xx:xx:xx] homeNetwork (-47) *
[D][WiFiMulti.cpp:151] run():        2: [9][xx:xx:xx:xx:xx:xx] REDACTED (-77) *
[D][WiFiMulti.cpp:151] run():        3: [11][xx:xx:xx:xx:xx:xx] REDACTED (-84) *
[D][WiFiMulti.cpp:151] run():        4: [1][xx:xx:xx:xx:xx:xx] REDACTED (-87) *
[D][WiFiMulti.cpp:151] run():        5: [1][xx:xx:xx:xx:xx:xx] REDACTED (-88)  
[D][WiFiMulti.cpp:151] run():        6: [1][xx:xx:xx:xx:xx:xx] REDACTED (-89)  
[D][WiFiMulti.cpp:149] run():  --->   7: [1][xx:xx:xx:xx:xx:xx] homeNetwork (-89) *
[D][WiFiMulti.cpp:151] run():        8: [11][xx:xx:xx:xx:xx:xx] REDACTED (-92) *
[D][WiFiMulti.cpp:151] run():        9: [3][xx:xx:xx:xx:xx:xx] REDACTED (-93) *
[D][WiFiMulti.cpp:151] run():        10: [9][xx:xx:xx:xx:xx:xx] REDACTED (-94) *
[I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: xx:xx:xx:xx:xx:xx SSID: homeNetwork Channel: 11 (-47)
[E][WiFiMulti.cpp:184] run(): [WIFI] Connecting Failed.
.[D][WiFiMulti.cpp:195] run(): [WIFI] delete old wifi config...
[D][WiFiMulti.cpp:198] run(): [WIFI] start scan
....[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 1 - SCAN_DONE
[WiFi-event] event: 1
AP Not valid, Waiting 30 seconds before restart, press 'prg' button to defaults settings or 'rst' button to restart now
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10044
load:0x40080400,len:5900
entry 0x400806ac
Testing EEPROM Library
EEPROM Size: 2048 bytes
Current language: es
Connecting to Wi-Fi...
[I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: homeNetwork
[I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: MIWIFI2
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
[WiFi-event] event: 0
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
[WiFi-event] event: 2
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 1 - SCAN_DONE
[WiFi-event] event: 1
[I][WiFiMulti.cpp:114] run(): [WIFI] scan done
[I][WiFiMulti.cpp:119] run(): [WIFI] 5 networks found
[D][WiFiMulti.cpp:151] run():        0: [11][xx:xx:xx:xx:xx:xx] homeNetwork_2 (-58) *
[D][WiFiMulti.cpp:149] run():  --->   1: [11][xx:xx:xx:xx:xx:xx] homeNetwork (-59) *
[D][WiFiMulti.cpp:151] run():        2: [11][xx:xx:xx:xx:xx:xx] REDACTED (-80) *
[D][WiFiMulti.cpp:151] run():        3: [9][xx:xx:xx:xx:xx:xx] REDACTED (-91) *
[D][WiFiMulti.cpp:151] run():        4: [4][xx:xx:xx:xx:xx:xx] REDACTED (-92) *
[I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: xx:xx:xx:xx:xx:xx SSID: homeNetwork Channel: 11 (-59)
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 4 - STA_CONNECTED
[WiFi-event] event: 4
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:419] _eventCallback(): STA IP: 1.1.1.82, MASK: 255.255.255.0, GW: 1.1.1.1
WiFi connected to IP address: 1.1.1.82
[I][WiFiMulti.cpp:174] run(): [WIFI] Connecting done.
[D][WiFiMulti.cpp:175] run(): [WIFI] SSID: homeNetwork
[D][WiFiMulti.cpp:176] run(): [WIFI] IP: 1.1.1.82
[D][WiFiMulti.cpp:177] run(): [WIFI] MAC: xx:xx:xx:xx:xx:xx
[D][WiFiMulti.cpp:178] run(): [WIFI] Channel: 11
Desactivando timer mqtt
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 1 - SCAN_DONE
[WiFi-event] event: 1
SSID 0 - wifi (92%, -54 dBm)
SSID 1 - wifi_2 (92%, -54 dBm)
SSID 2 - NOS-E736 (40%, -80 dBm)
SSID 3 - MEO-F255F0 (18%, -91 dBm)
SSID 4 - Vodafone-2EB31B (16%, -92 dBm)
SSID 5 - NOS-13B7 (12%, -94 dBm)
Welcome to FreeDS
Hostname: freeds_9c9c
mDNS responder started
Relay 1 Forced Off
Relay 2 Forced Off
Relay 3 Forced Off
Relay 4 Forced Off
ets Jun  8 2016 00:22:57
```

As I said FreeDS worked without issues for months before I soldered the pin headers.
This network has 2 Access Points for the whole house coverage as can be seen in the wifi scan.
It does not help to shut down one of those keeping only one in the spectrum...
I also tested with a really easy and simple WPA password in a new SSID homeNetwork_2, but it does not help either...

From the Access Point logs, I mostly get:

`xx:xx:xx:xx:xx:xx@wlan1: disconnected, extensive data loss, signal strength -55` (MAC Address of the WiFi 32 Kit)

This make me suspect that somehow I might be introducing some kind of interference when soldering the pin headers...

Any help or hint would be very appreciated.

Thanks

pablozg, 2022-06-04:
Hi, Please try first the last beta version https://github.com/pablozg/freeds/releases/download/1.1.0_Beta/FreeDS_1.1.0010_Beta.zip, because the Rev 2 is obsolete, if you still having issues then write here or you can use the official telegram channel at: https://t.me/freeds_es.

Thanks.

luisbarreto, 2022-06-09:
Hi Pablo,

Thanks for taking time answering me, mostly because this hardly seem to be a FreeDS specific software problem by itself...
I flashed the latest firmware Release 1.1.0010 Beta but the issue remains. 
Considering this, the issue isn't probably not FreeDS related, so sorry for raising this issue here, you can close it anytime you want.

Even so I just want to share a little more on the subject.
The connection to the wifi network is poor and I consistently seed the message `disconnected, extensive data loss, signal strength -51` being logged on the Access Point logs. 

I've been testing this issue with 4 Heltec WiFi Kit 32 with same results regardless the FreeDS version (tested 1.0.7 Final Rev 2 and 1.1.0010 Beta) or the board itself - all 4 suffer from the same wifi connectivity issue and this seems to be mainly a hardware/antenna design or at some extent might need some work at the wifi library level...
For the record, I was running one of these boards with FreeDS 1.0.7 Final Rev 2, for more than 4 months, without soldering their pin headers (fed by micro USB), without any noticeable problem. As soon as I soldered the pin headers, the wifi connectivity problems started. 

Things already tested and their outcomes regarding the wifi connectivity of these boards:
Baseline for comprehension:
- 1 SSID with 2 APs (both Mikrotik haP AC2), basically wired connected bridges, to extend the wifi range;

- channels 1 and 11 used initially, one  on each AP. It seems that changing the AP on channel 11 to channel 6 might have improved (very) little - channel 11 seems more crowded by neighbors;
- change from 20/40 channel bandwidth to just 20 Mhz (only) does not seem to improve anything;
- change from G/N band to either G or N (only) does not seem to improve anything;
- powering from micro USB or +5V pin (low noise good quality adapter +5v 1A) doesn't seem to have any impact

Seems that the pin headers cause some kind of antenna or impedance problem that make this board very difficult to work with even on a relatively small house.
For contextual purposes, I have several wifi networked devices like ESP8266, Sonoff's and shellys, inside the walls, on a balcony at a higher distance from the nearest AP and none of these have had any connectivity issues for several years now.

A final question: would it be feasible to put together an ESP32 with a separate 0.96" OLED display for this purpose using the same codebase and libraries? 

Thanks.

closed 2022-09-14 by pablozg