pablozg/freeds #58: What does the 'Inverter/LoadWatts' topic mean?

dml33, 2022-07-01:
Hi.

I am trying to publish via MQTT the set of topics related to the "ICC Solar MQTT" inverter type, as stated in the FreeDS manual:

```
Para dispositivo "ICC Solar MQTT".- Los topics son:
"Inverter/GridWatts" // Potencia de red
"Inverter/MPPT1_Watts" // Potencia string 1
"Inverter/MPPT2_Watts" // Potencia string 2
"Inverter/MPPT1_Volts" // Tension string 1
"Inverter/MPPT2_Volts" // Tension string 2
"Inverter/MPPT1_Amps" // Corriente string 1
"Inverter/MPPT2_Amps" // Corriente string 2
"Inverter/PvWattsTotal" // Potencia solar
"Inverter/SolarKwUse" // Energía diaria solar
"Inverter/BatteryVolts" // Voltaje batería
"Inverter/BatteryAmps" // Corriente bateria
"Inverter/BatteryWatts" // Potencia batería
"Inverter/BatterySOC" // Nivel batería
"Inverter/LoadWatts" // Consumo actual
"Inverter/Temperature" // Temperatura Inversor
```

What is exactly the "Inverter/LoadWatts" topic about? What is it used for? 

Because, with that description, it would be equivalent to "Inverter/GridWatts", which is the current grid power (negative: taken from the grid; positive: exported to the grid).

Thanks for the clarification.

Best regards.

pablozg, 2022-07-04:
Hi, LoadWatts means the total amount of watts of the load connected to the inverted, if you have your whole house connected to the inverter, then you should put the total watts that the house is using.

GridWatts means the amount of watts exporting / importing from the grid.

dml33, 2022-07-05:
Thank you, Pablo.

Understood. It is the current total consumption of the house. In my case, since the inverter does not give it directly, I have to calculate it as (PvWattsTotal + BatteryWatts) - GridWatts.

Great!

closed 2022-09-14 by pablozg