pablozg/freeds #59: Toggle PWM without writting eeprom each time

jorgemiguel4, 2022-07-07:
I have been using freeds for a few weeks now coupled with home assistant and it's been great.  In Portugal we have a "quarter hour netmetering" period, to squeeze most of the solar energy sometimes I have to toggle PWM on/off every couple minutes or so.  My weblog looks like this:
11:45:00 - [Command] PWM control: ON
11:45:00 - [Eeprom] DATA SAVED!!!!
11:45:56 - [Command] PWM control: OFF
11:45:56 - [Eeprom] DATA SAVED!!!!
11:54:00 - [Command] PWM control: ON
11:54:00 - [Eeprom] DATA SAVED!!!!
11:54:06 - [Command] PWM control: OFF
11:54:06 - [Eeprom] DATA SAVED!!!!
11:55:02 - [Command] PWM control: ON
11:55:02 - [Eeprom] DATA SAVED!!!!
11:58:26 - [Command] PWM control: OFF
11:58:26 - [Eeprom] DATA SAVED!!!!

Is there a way to disable eeprom writes to preserve eeprom or is a non issue?

pablozg, 2022-07-07:
Hi,

It's not a problem because freeDS is receiving a manual instruction, so freeDS needs to save the actual state to avoid failure on a eventul power loss or reboot.

If you send the "pwm to shutdown" command, freeDS needs to know and store it, because if you restart or shutdown it, when it starts up it will be in the same operating mode before the restart or shutdown.

Maybe you should take a look at your hassio routine to avoid sending continuous on/off commands.

jorgemiguel4, 2022-07-07:
I already keep the current state to avoid some writes but it's hard to do better since I have to pause PWM to recover some energy after use of some devices that consume more than my solar production in the 15min interval. I would be nice to have a flag in the payload to make it permanent or volatile.

pablozg, 2022-10-04:
Hola, @jorgemiguel4, tu propuesta la he integrado para la siguente beta que publique, si quieres probarla únete al grupo de telegram: https://t.me/freeds_es y te la mando por privado.

El payload es así:
```
/*
	Payload
	0 PWM OFF Permanent
	1 PWM ON  Permanent
	2 PWM OFF Volatile
	3 PWM ON  Volatile
*/
```

Y el comando se envia a ``` nombre_de_host/cmnd ``` y en formato json: ``` {"command":"pwm","payload":"3"} ```

jorgemiguel4, 2022-10-04:
Muchas gracias @pablozg

svcabre, 2022-11-05:
Hola @jorgemiguel4 @pablozg, 

Esta programación es realmente interesante para alargar la vida del ESP32. Podrías añadir esta opción de Volatile o Permanent también para PWM AUTO/MAN

pablozg, 2022-11-09:
> Hola @jorgemiguel4 @pablozg,
> 
> Esta programación es realmente interesante para alargar la vida del ESP32. Podrías añadir esta opción de Volatile o Permanent también para PWM AUTO/MAN

Hola, efectivamente para la opción de cambio entre auto / man también está implementado, lo podrás probar en cuanto se publique la beta 13.

svcabre, 2022-11-12:
Hola @pablozg,

Acabo de actualizar mi ESP32 otra vez desde 0 usando el programarESP32.bat y he realizado las pruebas. 

La implementación funciona correctamente para PWM ON o OFF

[image]

Pero en cambio para PWM MAN o AUTO me sigue apareciendo [Eeprom] DATA SAVED!!!!

20:23:27 - [Command] PWM control: OFF
20:23:30 - [Command] PWM control: ON
20:23:32 - [Command] PWM control: OFF
20:23:33 - [Command] PWM control: ON
20:23:47 - [Command] PWM mode set to: AUTO
20:23:47 - [Eeprom] DATA SAVED!!!!
20:23:49 - [Command] PWM mode set to: MANUAL
20:23:49 - [Eeprom] DATA SAVED!!!!
20:23:51 - [Command] PWM mode set to: AUTO
20:23:51 - [Eeprom] DATA SAVED!!!!
20:23:52 - [Command] PWM mode set to: MANUAL
20:23:52 - [Eeprom] DATA SAVED!!!!

Estoy utilizando MQTT y desde NodeRed mando los siguientes comandos:

Para AUTO {"command":"pwmman","payload":"2"}
Para MAN  {"command":"pwmman","payload":"3"}