pablozg/freeds #60: Compile freeds

jorgemiguel4, 2022-08-03:
What's the easiest way to compile freeds? 

Thank you

pablozg, 2022-08-04:
Hi, The best way is use VSCODE + platformio, then open the folder and compile.

jorgemiguel4, 2022-08-04:
I'm trying to compile 1.1.0 beta but some libs are missing. Can you provide a package with everything?

In file included from lib/ArduinoJson/src/ArduinoJson/Collection/CollectionData.hpp:7,
                 from lib/ArduinoJson/src/ArduinoJson/Array/ArrayFunctions.hpp:7,
                 from lib/ArduinoJson/src/ArduinoJson/Array/ArrayRef.hpp:7,
                 from lib/ArduinoJson/src/ArduinoJson.hpp:17,
                 from lib/ArduinoJson/src/ArduinoJson.h:9,
                 from C:/Users/Jorge/Desktop/FreeDS Compile/freeds-1.1.0_Beta/freeds-1.1.0_Beta/src/FreeDS.ino:37:
lib/ArduinoJson/src/ArduinoJson/Namespace.hpp:8:10: fatal error: ArduinoJson/Polyfills/preprocessor.hpp: No such file or directory
 #include <ArduinoJson/Polyfills/preprocessor.hpp>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\heltec_wifi_kit_32\src\FreeDS.ino.cpp.o] Error 1

diyeasy-es, 2022-08-04:
Hello, I had the same issue, but if you download the repository PID from https://github.com/pablozg/freeds you will find the lib.zip (I used this one and its working)

jorgemiguel4, 2022-08-09:
I'm still getting this:

lib/esp32ModbusTCP/src/esp32ModbusTCP.h:30:10: fatal error: FreeRTOS.h: No such file or directory

jorgemiguel4, 2022-08-16:
Can you share the entire folder please? Thank you

pablozg, 2022-08-16:
Freertos is part of the main framework and platformio must install it. I can't send you as a lib.

You can try to compile with arduino ide, but you have to install the esp32 framework and the libs from arduino library manager.

diyeasy-es, 2022-08-18:
Yes,

I think that is a framework version issue. Try to downgrade till Platform Espressif 32 v3.5.0

I think that new frameworks has embebed FreeRTOS, so you don't need the .h library

jorgemiguel4, 2022-08-18:
Downgrade Platformio IDE?

diyeasy-es, 2022-08-18:
No, in platform, only you need downgrade the espressif 32  platfrom

[image]

jorgemiguel4, 2022-08-18:
espressif32 downgrade did the trick. thank you very much

jorgemiguel4, 2022-08-18:
> Freertos is part of the main framework and platformio must install it. I can't send you as a lib.
> 
> You can try to compile with arduino ide, but you have to install the esp32 framework and the libs from arduino library manager.
> 
> Be aware that the uploaded code is not the actual Beta 10 code, and I don't know when will be available.

Can you share the new code here?

pablozg, 2022-09-14:
The new code will be available when it changes its status from beta to stable.

Could you say me what changes wants to do in the code?

jorgemiguel4, 2022-09-14:
Like a said in the other thread I started, I want to disable eeprom writes for pwm and pwmman mqtt commands. The way it's implemented in freeds it rewrites all the configuration everytime it receives a command and I'm  concerned about the longevity of ESP32 in my use case.

closed 2022-10-12 by pablozg