pablozg/freeds #65: Distinct Temperature Sensor for Automatic and Manual mode

gabrielvessteves, 2022-09-22:
Hello,
it would be a nice feature to set different Turn On / Turn Off temperatures to Manual and Automatic mode.
As an example:
**At night i set:**
Turn on temperature= 35ºC
Turn off temperature= 40ºC
Apply to Manual mode.
Programador Horario:
Time start: 4:00AM
Time Stop: 7:00PM

This way, i always have hot water to the morning bath.

But it would be very nice if **during the day** i could get water until 70ºC .
So i can see it could work if i set:
 Turn off temperature = 70ºC
 Apply to Auto mode.

This way it could improve efficiency of the home system.

Thanks for the excellent work on this software.

pablozg, 2022-10-12:
Hi, I will add it to the TODO list.

Csontikka, 2024-10-19:
> Hi, I will add it to the TODO list.

Dear pablozg!

I've also encountered this issue and came to the conclusion that the solution proposed here would address the problem effectively. I kindly ask you to prioritize this feature, as it was originally requested in 2022, and there has been no progress on it since then. It would be incredibly helpful for me and likely many others. Thank you for your attention and hard work!

pablozg, 2024-10-20:
Hola, en aquella fecha no se disponia del parametro de temperatura mínima, que ahora cubre esa necesidad, con ella se mantiene siempre la temperatura que desees, una vez logrado si dispone de excedentes seguirá calentandola hasta la temperatura de corte indicada en el modo automático o según la programación en el modo manual.