pablozg/freeds #66: Salida 25 para PWM

manunope, 2022-09-23:
Hola, soy @manunope.
Quiero utilizar un ESP8266 o ESP32 de tamaño reducido y sin pantalla (siempre uso el movil)
La pregunta es en que parte del código debo cambiar la salida PWM (25), las salidas 12, 13, 14 y 27,  la entrada 2 (temperatura), la entrada 0 (pulsador), en fin, adaptar las IOs al tipo de micro que vaya a utilizar.
Gracias.

pablozg, 2022-10-04:
Hola, lo primero decirte que no puedes usar un esp8266 puesto que el código está pensado para un eps32, el código subido actualmente se corresponde con la 1.0.7 rev2, y no con las últimas betas, por lo que si necesitas las funcionalidades de las últimas versiones no te valdrá.

Por otra parte, casi la totalidad de las definiciones vienen en el fichero freeds.ino.

manunope, 2022-10-04:
Entendido. Muchas gracias.

closed 2022-10-04 by manunope