pablozg/freeds #67: GoodWe DNS

edurnebernal, 2022-10-10:
Hi,
Are you planning to add support for GoodWe DNS inverters? 
I have seen that ES/EH/EN protocols are already implemented, but these don't seem to work with my GoodWe DNS .

Thanks.

pablozg, 2022-10-12:
Hola, tienes información sobre la comunicación que usan los GoodWe DNS? Si la tienes pásamela y en cuanto pueda la intento integrar.

¿Podrás hacer la prueba de funcionamiento?, para una comunicación más fluida únete si no lo estás ya al grupo oficial de telegram en https://t.me/freeds_es.

edurnebernal, 2022-10-15:
Esta es la información que he podido encontrar respecto al protocolo de comunicación de la familia GoodWe DT ( a la que parece pertenecer los inversores GoodWe DNS): [https://github.com/MiG-41/Modbus-GoodWe-DT](https://github.com/MiG-41/Modbus-GoodWe-DT) y [https://github.com/marcelblijleven/goodwe](https://github.com/marcelblijleven/goodwe)

> ¿Podrás hacer la prueba de funcionamiento?, para una comunicación más fluida únete si no lo estás ya al grupo oficial de telegram en https://t.me/freeds_es.
Sí, podría hacer pruebas de funcionamiento sin problema.

closed 2023-01-05 by pablozg