pablozg/freeds #68: Shelly EM - Make it work with just one CLAMP

jaldwt, 2022-10-11:
Hello, pablozg :)

I tried to make it work with one CLAMP, but I'm not able to do it.
Where i have the EM meter is far away form the solar unit, so i cant place the second clamp on the production side. 
Is there configuration that able me to do it? Or even do it by software?

Thank you, 
Best regards, 
Jaldwt

pablozg, 2022-10-12:
Hi, you can use only one clamp, the only needed clamp is the number 1 and it must be attached to the grid wire.

The second clamp is only for solar production information, and is not needed.

You must put the clamp to have positive values (green color) when you export energy to the grid, and negative (red color) when you import from the grid.

For a quick support please use the telegram group: https://t.me/freeds_es

jaldwt, 2022-11-02:
I really cant access to the group i was banned and i dont know why(if you can, pls unlock me, i have same user name as here). 
At this moment my Shelly when is exporting gives negative values and i have the GEN3 ulktra low board. Can someone leave an explication of the pwm configuration to this hardware set?  Thank you :)

pablozg, 2022-11-02:
Hi, you must get positive values when you are exporting to the grid, you can read the manual here: https://docs.google.com/document/d/1-XLCqHEbpqEPi4geI4CbANmGxodExC__zDvND89xjIU/edit?usp=sharing

About the group I can't find your nick as banned, can you say me what name have in telegram? Thanks.

jaldwt, 2022-11-02:
> Hi, you must get positive values when you are exporting to the grid, you can read the manual here: https://docs.google.com/document/d/1-XLCqHEbpqEPi4geI4CbANmGxodExC__zDvND89xjIU/edit?usp=sharing
> 
> About the group I can't find your nick as banned, can you say me what name have in telegram? Thanks.

Yes, sure, search for André Dias.

pablozg, 2022-11-02:
Now you can enter to the group