pablozg/freeds #69: FreeRtos Error during compiling code

ahmadsadiq40, 2022-11-15:
Hello sir i hope you're doing well.
Actually i want to discuss my issues with you about source code.
When i build the code i got freertos errors, some libraries. Can you please help me ?

Linux-RISC, 2022-11-21:
Same problem here:
1. FreeRTOS.h not found. I can download this file and copy it to include directory, but then a second problem appears:
2. crypto/base64.h not found

I have tried compiling under Windows and Debian GNU/Linux, but this error is always retrieved.

Spanish note:
He intentado resolver este problema pero siempre me da el mismo error, ¿hay alguna manera de resolverlo?

pablozg, 2022-11-22:
> Hello sir i hope you're doing well. Actually i want to discuss my issues with you about source code. When i build the code i got freertos errors, and crypto/base64 error. Can you please help me ?

Hi,

1º Install VSCodium and pyhton
2º Download cpptools and platformio from: https://marketplace.visualstudio.com/items?itemName=ms-vscode.cpptools and https://marketplace.visualstudio.com/items?itemName=platformio.platformio-ide.
3º Install the extensions using the option "Install from VSIX"

Download the source code again (I have fixed the compile issues) and try to compile it.

Linux-RISC, 2022-11-23:
This works for me:

Development system installation:
- install VSCodium (optional, you can use any editor, https://github.com/VSCodium/vscodium/releases)
- install Python (https://www.python.org/downloads/release/python-3110/)
- install platformio: python get-platformio.py, download from https://raw.githubusercontent.com/platformio/platformio-core-installer/master/get-platformio.py
- install git (https://git-scm.com/download/win)
- download source code: git clone https://github.com/pablozg/freeds.git
- compilation: enter into freeds directory and run "pio run"
- generate the file system: enter into freeds directory and run "pio run --target buildfs"

Tested on Debian GNU/Linux 11.5 amd64:
$ sudo apt install python

$ wget https://raw.githubusercontent.com/platformio/platformio-core-installer/master/get-platformio.py

$ sudo apt install python3-venv

$ python3 get-platformio.py

$ sudo apt install git

$ git clone https://github.com/pablozg/freeds.git

$ cd freeds

Compilation:
$ $HOME/.platformio/penv/bin/pio run
$ $HOME/.platformio/penv/bin/pio run --target buildfs

You can also modify the file .profile of your user:
PATH=$PATH:$HOME/.platformio/penv/bin

and compile:
$ pio run
$ pio run --target buildfs

closed 2022-11-25 by ahmadsadiq40