pablozg/freeds #70: Adding support to Goodwe XS series

gomezguillermo, 2022-12-01:
Hola he estado probando FreeDS con el inversor Goodwe GW3000-XS y no logro hacer que funcione, noto que, haciendo tcpdump de los paquetes UDP que entran y salen del inversor, secuencia para pedir datos a este inversor es diferente en vez de la secuencia que esta para los modelos EH/ES/EM dentro de freeds.

Actualmente freeds envía en formato HEX: `aa 55 c0 7f 01 06 00 02 45`, sin tener una respuesta del inversor.

Ahora, si envío  `7f 03 75 94 00 49 d5 c2`, si tenemos una respuesta valida, por ejemplo:

`aa557f0392160c010f0f0d083e002200000000ffffffffffffffffffffffffffffffffffffffffffff0938ffffffff001effffffff1389ffffffff000002d00001000000000000ffffffffffffffffffffffffffffffff016fffffffff004c0000021e0000004c00200000ffff0000ffff0000ffff0000ffffffffffffffffffff05680efdffffffffffff001dffffffffffff01f00046d173`

He intentado compilar el código de freeds con estos cambios y obtengo un firmware.bin ligeramente mas liviano que el zip oficial pero no logro hacerlo arrancar, sigo intentando.

Si alguien tiene tiempo de probar o generar un firmware.bin se lo agradecería.

Un saludo

pablozg, 2022-12-01:
Tienes algún sitio donde ver que bytes corresponden a que dato?, así podría añadir un nuevo modelo con esa petición, pero necesito saber a que corresponde cada cosa.

El código colgado corresponde a la 1.0.7 rev2, los binarios actuales corresponden a la beta 14, cuyo código no es público por el momento.

gomezguillermo, 2022-12-01:
En esta librería de python (https://pypi.org/project/goodwe/) manejan la respuesta para los modelos de GoodWe ET, EH, BT, BH, ES, EM, BP, DT, MS, D-NS, y XS en lo que tenga un poco de tiempo me pongo a descifrar los bytes

pablozg, 2022-12-04:
> f 03 75 94 00 49 d5 c2

Hola, si no estás en el grupo de FreeDS en telegram (https://t.me/freeds_es), únete y te paso una versión de prueba por privado, mandame primero un mensaje para saber que eres tú.

Tienes baterías instaladas?

gomezguillermo, 2022-12-04:
si estoy en el grupo freeds puedes enviarlo a @guillegomez y lo pruebo. No tengo baterías.

closed 2023-01-05 by pablozg