pablozg/freeds #71: Añadir soporte para Enphase Envoy

jmatiasGH, 2022-12-16:
Es posible obtener información del aparato (envoy) que monitoriza los microinversores de Enphase sin tener otros dispositivos intermediarios, en Home Assistant hay una integración ya hecha por lo que es ver como lo hace él (https://github.com/vk2him/Enphase-Envoy-mqtt-json/blob/main/envoy_to_mqtt_json.py). De ahí se podría ver que datos se sacan, como se conecta y ver como adaptarlo al código del FreeDS. Se trata de una web que sirve un stream de datos cada segundo.
Para colaborar me he instalado el VScode con la extensión Platformio, he conseguido compilar un firmware, parto de cero para colaborar como pueda. Estoy pendiente de un ESP32 para ir haciendo pruebas modificando el firmware original.
Me ofrecí a compartir acceso a mi monitor envoy para hacer pruebas, obviamente no lo pongo aquí para no compremeter la seguridad de mi equipo, pero mi ofertar sigue abierta si va a suponer una mejora para todos los que tenemos Enphase.
Estoy en el grupo de Telegram y 8 personas montamos otro grupo derivado para ver si sacamos esto adelante entre todos.

pablozg, 2022-12-17:
Hola, prueba a escribir esta url en un navegador y me comentas si responde correctamente, y en ese caso me copias la salida para ver que datos envia.

http://ipdelenvoy/stream/meter

Saludos.

jmatiasGH, 2022-12-17:
Al conectarme saca un bloque como este cada segundo, estos valores son de ahora que es de noche. A ver si mañana te mando un ejemplo con producción solar:

data: {"production":{"ph-a":{"p":-2.199,"q":130.71,"s":137.243,"v":232.698,"i":0.59,"pf":0.0,"f":50.0},"ph-b":{"p":0.0,"q":0.0,"s":0.0,"v":0.0,"i":0.0,"pf":0.0,"f":0.0},"ph-c":{"p":0.0,"q":0.0,"s":0.0,"v":0.0,"i":0.0,"pf":0.0,"f":0.0}},"net-consumption":{"ph-a":{"p":262.63,"q":-225.749,"s":359.985,"v":232.853,"i":1.545,"pf":0.71,"f":50.0},"ph-b":{"p":0.0,"q":0.0,"s":0.0,"v":0.0,"i":0.0,"pf":0.0,"f":0.0},"ph-c":{"p":0.0,"q":0.0,"s":0.0,"v":0.0,"i":0.0,"pf":0.0,"f":0.0}},"total-consumption":{"ph-a":{"p":260.431,"q":-356.459,"s":496.965,"v":232.776,"i":2.135,"pf":0.52,"f":50.0},"ph-b":{"p":0.0,"q":0.0,"s":0.0,"v":0.0,"i":0.0,"pf":0.0,"f":0.0},"ph-c":{"p":0.0,"q":0.0,"s":0.0,"v":0.0,"i":0.0,"pf":0.0,"f":0.0}}}

jmatiasGH, 2022-12-17:
El bloque de "production" tiene los parámetros de producción, potencia "p", voltaje "v", intensidad "i", frecuencia "f"... Y otros que no he identificado ahora mismo. La p es negativa por el pequeño consumos de los microinversores y el monitor.
La net-consumption es la diferencia entre lo importado de la red y lo exportado.
El total-consumption es como su nombre dice el consumo total de la casa.

jmatiasGH, 2022-12-17:
Creo que unas de las variables son las siguientes:
p(kW) -- Potencia útil.
q(kVAR) -- Potencia campos. ?
s(kVA) -- Potencia aparente. ?
Pf -- Factor de potencia.

https://www.electricaplicada.com/que-es-la-potencia-aparente-kva/

pablozg, 2022-12-18:
Cual es la longitud del password generado?

jmatiasGH, 2022-12-18:
> Cual es la longitud del password generado?

8 caracteres alfanuméricos con mayúsculas y minúsculas.

pablozg, 2022-12-18:
> > Cual es la longitud del password generado?
> 
> 8 caracteres alfanuméricos con mayúsculas y minúsculas. Usuario "installer".

Necesitaría acceso al inversor para probar la implementación, contacta conmigo en el canal de telegram y vemos como hacerlo https://t.me/freeds_es

jmatiasGH, 2022-12-18:
> > > Cual es la longitud del password generado?
> > 
> > 
> > 8 caracteres alfanuméricos con mayúsculas y minúsculas. Usuario "installer".
> 
> Necesitaría acceso al inversor para probar la implementación, contacta conmigo en el canal de telegram y vemos como hacerlo https://t.me/freeds_es

A poder abro de nuevo el puerto y te aviso diciendote a donde conectar. Ahora estoy fuera, en seguida que pueda te digo cosas. Gracias Pablo!!

jmatiasGH, 2022-12-18:
He habilitado de nuevo el puerto para que puedas acceder al Enphase Envoy Monitor. No he podido hacer una captura de día en producción porque he estado todo el día fuera de casa. Voy a ver si te puedo pasar la información de conexión por un privado en Telegram ahora.

jmatiasGH, 2022-12-27:
Informar que he comenzado a hacer las pruebas con el GEN8PRO de Francisco con carga real y ha mejorado mucho el funcionamiento con la conexión directa por Envoy de su comportamiento. Hay que pulir la configuración y ver esos parámetros de voltaje e intensidad de la producción que no nos cuadran. Voy a intentar hablar con los instaladores para que me revisen porque puede ser y ver si es algún tipo de fallo en la instalación o ajuste por software que esté falseando los datos.

closed 2023-01-05 by pablozg