pablozg/freeds #73: Añadir control de temperatura para el Triac

evaristocuesta, 2022-12-31:
Estaría bien añadir un control de temperatura para el sensor del Triac, al igual que ya existe uno para el sensor del termo. 
[image]
La idea sería añadir duplicar los controles ya existentes en el termo para el triac y así poder configurar de forma independiente la temperatura de encendido y apagado del termo y el triac. 
Ahora mismo, Francisco Ureña de SolarTV aconseja poner el sensor del triac en el del termo y así desconectar el gestor en caso de que el triac alcance la temperatura indicada y así alargar la vida del gestor. Si se desarrolla esta funcionalidad, podría hacerse de forma independiente y tener control para el termo y el triac. 
En caso de que os parezca interesante, si os parece bien, podría intentar añadir esta funcionalidad.

pablozg, 2023-01-05:
Gracias por tu sugerencia, por el momento no es una prioridad, ya que si la disipación está bien dimensionada no es necesario hacer ningún tipo de control sobre el triac, 

Mantendré abierta la solicitud por si en un futuro se implentara.

evaristocuesta, 2023-01-05:
Perfecto. Gracias!

La sugerencia la hice porque le compré el gestor de excedentes a SolaresTV y te recomienda que lo pongas con el sensor de triac para controlar la temperatura y alargar la vida del gestor. 
Hasta ahora no me ha pasado de 27º aunque sólo lo estoy usando de prueba con un calefactor de 800 W. A ver a cuánto se pone con el termo de 2300 W.

closed 2025-11-23 by pablozg