pablozg/freeds #74: Añadir más franjas horarias para el programador horario

evaristocuesta, 2022-12-31:
Tener al menos tres franjas horarias en lugar de una para el programador horario. De esa manera podríamos decir por ejemplo que se ponga en manual de 7:00-8:00, de 15:00-17:00 y de 22:00-23:00, por poner un ejemplo. 
[image]
También me ofrezco para intentar hacerlo en caso de que os parezca interesante.

pablozg, 2023-01-05:
Hola, el programador horario lo tengo pensado desde hace tiempo, pero no he tenido tiempo de ponerme con ello, quiero que sea versatil al estilo como lo hace tasmota, pudiendo elegir el tipo de acción [ON, OFF, MANUAL, AUTO] según el tipo de objetivo, y poder elegir las salidas a las que aplicarlo, ya sea el propio derivador o las salidas de relés.

Tengo que ponerme a ello, pero todavía no he encontrado tiempo.

Pero en el apartado web si que podrías ayudar y mejorarlo si te apetece, pero ten en cuenta que la memoria disponible para ella son 896Kb.

evaristocuesta, 2023-01-05:
Estaría muy bien el programador con elección de tipo de acción. 

Sobre la web, soy más de backend que de frontend, pero bueno, dime qué es lo que tienes pensado e intento echarte una mano.