pablozg/freeds #76: Integración con Sofar Solar HYD-5000-EP y otros modelos Sofar Solar HYD

svcabre, 2023-02-08:
Hola Pablo,

Se podría ampliar en una nueva versión la integración con Inversores nuevos de Sofar Solar?

Los siguientes registros Sofar Solar HYD-5000-EP son:

0x0485 -> ActivePower_OutputTotal 
0x04AF -> ActivePower_LoadSys 
Restando estos dos registros se obtiene el valor potencia de red :
0x04AF - 0x0485 = Inverter/GridWatts

0x0584 -> Voltage_PV1 -> Inverter/MPPT1_Volts
0x0585 -> Current_PV1 -> Inverter/MPPT1_Amps
0x0586 -> Power_PV1 -> Inverter/MPPT1_Watts
0x0587 -> Voltage_PV2 -> Inverter/MPPT2_Volts
0x0588 -> Current_PV2 -> Inverter/MPPT2_Amps
0x0589 -> Power_PV2 -> Inverter/MPPT2_Watts

0x05C4 -> Power_PV_Total -> Inverter/PvWattsTotal

0x0604 -> Voltage_Bat1 -> Inverter/BatteryVolts
0x0605 -> Current_Bat1 -> Inverter/BatteryAmps
0x0606 -> Power_Bat1 -> Inverter/BatteryWatts

0x0608 -> SOC_Bat1 -> Inverter/BatterySOC

0x041A -> Temp_HeatSink1 -> Inverter/Temperature

svcabre, 2023-02-08:
[SOFAR-G3 External Modbus Protocol-Engilsh-V211103.xlsx](https://github.com/pablozg/freeds/files/10688549/SOFAR-G3.External.Modbus.Protocol-Engilsh-V211103.xlsx)

pablozg, 2023-02-11:
> Hola Pablo,
>
> Se podría ampliar en una nueva versión la integración con Inversores nuevos de Sofar Solar?
>
> Los siguientes registros Sofar Solar HYD-5000-EP son:
>
> 0x0485 -> ActivePower_OutputTotal [...]

Que protocolo de modbus usa, el RTU o TCP, otra cuestión cual es el id por defecto 0 o 1?

svcabre, 2023-03-21:
Hola Pablo,

Es RTU el ID por defecto es 1 pero puede configurase en el inversor por otro ID distinto.