pablozg/freeds #77: Conexion Envoy Enphase

jmercadofraguasVillalba, 2023-02-13:
He probado a conectarlo con envoy enphase y no recoge datos, ¿puede ser que enphase haya cambiado la forma de conectarse?,  He probado a conectarme por api mediante postman y si que tira.

pablozg, 2023-02-14:
Hola, has seguido los pasos del manual de usuario? 

Están en la página 65 en adelante.

[Manual](https://docs.google.com/document/d/1-XLCqHEbpqEPi4geI4CbANmGxodExC__zDvND89xjIU/edit?usp=sharing)

jmercadofraguasVillalba, 2023-02-14:
Pablo, lo primero de todo muchisimas gracias por el interés.

Si he seguido los pasos del manual correctamente y he repetido el proceso
varias veces.

Informándome un poco, creo que el envoy ya no permite la conexión que está
realizando freeds, y ahora requieren de token de autorización previo a
reliazar la conexión.

He intentado realizar la conexión a través de un script de python con las
credenciales de installer y  me devuelve un error de certificado. [ssl:
certificate_verify_failed], creo que puede ser algo parecido a lo que
ocurre en freeds, no se si hay algún tipo de log en freeds que me devuelva
más info.

También he leido por algunos foros que el usuario para los envoy nuevos
podría ser envoy y no llevar pass, la verdad que ahora mismo me encuentro
un poco perdido sobre como hacerlo funcionar, pero seguiré investigando.

Si por tu lado encuentras algún avance, por favor, házmelo saber.

muchisimas gracias.

jmatiasGH, 2023-04-01:
Hola @jmercadofraguasVillalba! Puede ser que esté intentando capturar los datos de una de las fases donde no estés monitorizando corriente? O no tiene nada que ver? Ayudé a @pablozg a hacer la integración con el mío que es una instalación monofásica, pero deduzco que en el stream de datos está configurado para coger una de las fases que puede monitorizar. Quizás lo tengas conectado en una que no esté siendo la monitorizada. O directamente te sale errores al intentar conectar? Lo suyo sería saber si puedes ver el stream de datos y si puedes ver qué datos te devuelve y si queréis comparamos con el mío.
Quería proponerle a @pablozg que al configurar la conexión del envoy se pudiera elegir una fase o la suma de ellas, para los que tengan trifásico que ya ha habido algún caso que pedía por su caso con una instalación trifásica.

jmatiasGH, 2023-04-01:
@jmercadofraguasVillalba si introduces está URL en un navegador, con la IP de tu envoy y la clave de acceso que has obtenido que te sale? 
http://ipdelenvoy/stream/meter
Toca salir un Stream que da datos cada segundo. Si se analiza lo que sale se puede observar que hay tres fases. Por defecto está para trabajar con la phA.

marcoisel, 2024-02-03:
Hola Pablo, primero que nada gracias y enhorabuena por tu excelente trabajo.

Conecté mi enviado D5.0.62(5e57a9) con tres fases y lo configuré como tal, pero la producción solar que se muestra es solo de una fase, creo que L1. El consumo de red es correcto.

Gracias,
Marco
[image]
[image]