pablozg/freeds #78: ESP32-S3

Rsalvador2008, 2023-02-16:
is this compatible with the freeds? how?

Amazon only send me the esp32 v3 (esp32-s3)

Mizuba74, 2023-02-28:
I'm having the same problem, do you found a solution?
Thanks!

Rsalvador2008, 2023-02-28:
I have swapped my esp32 s3 for the normal esp32-d0

-------- Mensagem original --------

Data: 28/02/23 12:51 (GMT+00:00)
Para: pablozg/freeds ***@***.***>
Cc: Rsalvador2008 ***@***.***>, Author ***@***.***>

Mizuba74, 2023-02-28:
Do you by any chance have a link for the esp32-d0?

Rsalvador2008, 2023-02-28:
I have bought it in Portugal in the following store

https://www.botnroll.com/pt/arduino-controladores/2985-esp32-com-display-oled-128x64.html

Check with vendor first before place the order
-------- Mensagem original --------

Data: 28/02/23 12:54 (GMT+00:00)

Goaheadz, 2023-06-20:
Any update how to program esp32-s3?
Its getting hard to find old versions.
I have bougth one and dont want to return it.

Goaheadz, 2023-06-21:
I tried to compile the source to esp32 V3, but theres alot of standard libraries missing.
It would not be hard for pablozg, to compile this for S3-V3, using "visual code source" free software.

Already sent back my esp32-S3-V3 to seller...Now I dont know what to buy :(

JohnnyPleasures, 2023-06-30:
https://www.amazon.es/-/pt/dp/B07RWHHJXT/ref=cfb_at_prodpg

Used this One. Works great using shelly EM

joanorsky, 2023-09-05:
So.. no solution for the S3 then?

pablozg, 2023-09-06:
> I tried to compile the source to esp32 V3, but theres alot of standard libraries missing. It would not be hard for pablozg, to compile this for S3-V3, using "visual code source" free software.
> 
> Already sent back my esp32-S3-V3 to seller...Now I dont know what to buy :(

The problem with S3 is this chip version lost some need functionalities as Pin 25 and the analog output, so it's not a good replacement for esp32.

Rsalvador2008, 2023-09-06:
Just order the regular esp32 v1 version you should find some on Google, search for esp32 v1

Ricardo Salvador

-------- Mensagem original --------

Data: 06/09/23 21:01 (GMT+01:00)

rjsbandeira, 2024-01-27:
Same problem here. Amazon send me the esp32-s3 version :/ 
any update?

Jocaberto, 2024-03-03:
Hello, I also have the same problem, is there any solution for ESP-S3?

AmshaSharam, 2025-06-25:
settings for  ESP32-S3-WROOM-1-N16R8 
i didnt try with this code. 

On platformio.ini :
; Flash: 16MB QD, PSRAM: 8MB OT
[env:esp32-s3-devkitc-1-n16r8]
platform = espressif32
board = esp32-s3-devkitc-1-n16r8v
framework = arduino

board_build.arduino.memory_type = qio_opi
board_build.flash_mode = qio
board_build.psram_type = opi
board_upload.flash_size = 16MB
board_upload.maximum_size = 16777216
board_build.partitions = default_16MB.csv
board_build.extra_flags =
-DBOARD_HAS_PSRAM

monitor_speed = 115200
monitor_port = COM4
upload_port = COM4
upload_speed = 921600

pablozg, 2025-06-27:
No es un problema de configuración en platformio, es un problema de hardware, el ESP32-S3 no dispone de pin 25 que es el necesario para controlar las cargas, por lo que no es un reemplazo directo.

IvanSanchez, 2025-06-27:
@pablozg Pero el S3 (incluídas los devkits de S3) tiene GPIOs de sobra (45 en total). Se podría definir el GPIO 25 para ESP32 y, por ejemplo, el GPIO 21 para ESP32S3. Es cuestión de pelearse con los `#ifdef`s y las variables de entorno que meta platformio.

Hasta donde alcanzo a ver, el factor de forma de los devkits de ESP32S3 es distinto de los devkits de ESP32, así que de todas maneras habría que diseñar una PCB distinta.