pablozg/freeds #82: Exponer fecha de compilación e ID único a través de HTTP+JSON

IvanSanchez, 2023-04-07:
Buenas,

Estoy haciendo experimentos para una potencial integración con homeassistant (ahora mismo no puedo prometer nada).

Para poder hacer una buena integración, sería necesario poder pedir a FreeDS algunos datos como:
- Versión de FreeDS
- Fecha de compilación
- Algún ID único (p. ej, los bytes de la MAC que forman parte del nombre de `config.hostServer`)
- (opcionalmente) Dirección MAC completa
- La cadena de texto literal `FreeDS` en algún sitio

El objetivo es poder comprobar de manera programática si estoy hablando con un FreeDS o no (y disparar un error si se usa una versión de FreeDS no soportada, etc).

Creo que la manera más práctica sería exponer esos datos como parte del JSON devuelto por `http://$ip/masterdata`, p. ej,

```
{
"FreeDSVersion": "1.7.0",
"FreeDSCompilationDate": "2023-03-01",
"ID":"abcd",
"wversion":25,
"PwmMaster":30,
"tempShutdown":0,
    (.....etc.....)
}
```

pablozg, 2023-04-08:
Hola, el json al que te refieres solo está disponible en la 1.0.7 rev 2, en las ultimas betas ya no es válido.

En las betas puedes solicitar la información en http://ipdelfreeds/api/common y de devuelve un json como este:

{"title":"FreeDS (freeds_0200)","version":"Pre 1.1.0016 Beta Build: 02042023 Web: 1.1.3","workingmode":"Shelly EM","date":"Fecha: 08/04/2023 Hora: 17:26:52","uptime":"Uptime: 6d 05:20:26"}

si es necesario podria añadirle la mac del esp32 para poder diferenciarlo.

IvanSanchez, 2023-04-08:
@pablozg Ah, genial, pues entonces me lo apunto, y lo despacho cuando tenga algo de tiempo libre. El poder saber la versión de FreeDS instalada me vendrá bien.

He estado viendo que sería posible leer la MAC a través de python (vía https://pypi.org/project/getmac/ ); si hago experimentos y funcionan (la MAC que obtengo a través de python coincide con la ID que FreeDS se asignaa  sí mismo), entonces no sería necesario exponer esta información.

Más que la MAC, sería útil exponer el `hostname` del FreeDS. Supongo que habrá gente que configurará el `hostname` a algo distinto de `FreeDS ABCD`, y la integración de HASS debería ser capaz de usar ese nombre.

Por cierto: https://gitlab.com/IvanSanchez/homeassistant-freeds

pablozg, 2023-04-10:
> Más que la MAC, sería útil exponer el `hostname` del FreeDS. Supongo que habrá gente que configurará el `hostname` a algo distinto de `FreeDS ABCD`, y la integración de HASS debería ser capaz de usar ese nombre.

La parte entre parentesis de "title":"FreeDS (freeds_0200)" corresponde al host configurado por cada persona.

IvanSanchez, 2023-04-10:
No soy un experto en home assistant, así que no me conozco toda la documentación todavía. Pero ayer estuve leyendo https://developers.home-assistant.io/docs/config_entries_config_flow_handler/#unique-id-requirements , y ahí dicen:

> #### Example acceptable sources for a unique ID
> - MAC address: [...] Only obtain the MAC address from the device API or a discovery handler. Tools that rely on reading the arp cache or local network access such as `getmac` will not function in all supported network environments and are not acceptable.

> #### Unacceptable sources for a unique ID
> - Hostname if it can be changed by the user

Es decir, que yo puedo *suponer* que el nombre de host contiene una parte única; pero como es configurable no es una muy buena idea.

Probé a implementar `getmac`, y por desgracia no funciona como esperaba (además, hay usuarios que tienen el FreeDS detrás de un router con NAT).

Lo ideal sería poder tener la MAC completa en `/api/common`. Las otras opciones sencillamente no tienen la fiabilidad suficiente.