pablozg/freeds #86: Nuevo modo de funcionamiento: Home Assistant

IvanSanchez, 2023-04-15:
Después de haber estado trasteando bastante con HA, he visto que tiene una interfaz de WebSockets documentada: https://developers.home-assistant.io/docs/api/websocket/

De aquí sale la idea de poder configurar FreeDS como *cliente* de HA. No para enviar los datos a HA, sino para **leer** datos de HA.

La idea general es que FreeDS se conecte como cliente de websockets a HA según su API, y se suscriba a los cambios de *entidades* (con `subscribe_entities`, como hace su interfaz web). Entonces, HA empezaría a mandar todos los cambios a FreeDS. FreeDS deberá *filtrar* esos eventos, para quedarse únicamente con las entidades que representen la potencia del inversor y de la red. (Que yo sepa, no es posible pedir a HA que mande los cambios de entidades concretas)

Aparte supongo que habría que cambiar la página de configuración para que el usuario pueda decidir qué entidades de HA habría que usar (de manera similar a configurar *topics* de MQTT; al fin y al cabo son cadenas de texto).

Supongo que implementar esta funcionalidad es un cambio sustancial, pero no es urgente. Creo que sí es interesante de cara a poder soportar más dispositivos sin tener que implementar el cliente en FreeDS directamente.

pablozg, 2023-04-16:
De momento ha de quedarse en standby, ahora mismo no tengo tiempo para ponerme a levantar hassio para hacer pruebas, pero lo dejo abierto para un futuro.

IvanSanchez, 2023-05-01:
Si quieres te puedo montar una instancia de hassio para que hagas pruebas. Sólo tienes que avisar.

frapersan, 2023-05-02:
Yo actualmente cojo para freeds la información de HA. Tengo configurado varios disparadores (en función de las variables que quiero informar) que provocan el envío a través de MQTT. Si necesitas alguna información o ayuda estoy a vuestra disposición.

pablozg, 2023-05-03:
Hola, en estos momentos estoy bastante liado y no tengo apenas tiempo, ese cambio requiere tiempo y cambios en la estructura de almacenaje de datos y de funcionamiento, además de ajustar el apartado web.

En un futuro si tengo más tiempo se puede mirar de integrar, de todas formas estamos hablando de un número fijo (máximo) de topics o sin embargo es a libre elección.

La forma más sencilla sería saber el topic a suscribir y que en hassio decidas que topic enviar o no.

IvanSanchez, 2023-05-03:
Ojo, que mi idea es que FreeDS lea el flujo de websockets de HA. Eso es **distinto** de usar MQTT (entre otras cosas, MQTT va por puerto TCP 1833 y websockets va por el 80). El público objetivo es gente que tenga FreeDS y HA pero **no** tenga un broker de MQTT montado (¡porque no hace falta!).

---

Otra opción que podemos considerar es que yo cambie https://gitlab.com/IvanSanchez/homeassistant-freeds para que HA pueda enviar datos **activamente** a FreeDS. No es lo ideal: me gustaría que FreeDS fuera capaz de leer datos de un A "vainilla", a pelo; pero por otra parte creo que hay bastantes usuarios que tienen puesta mi integración

Para esta segunda opción creo que es posible reutilizar el protocolo de FreeDS maestro/esclavo: desde HA levanto un servidor web que exponga `/masterdata`, HA se hace pasar por un FreeDS en modo maestro, el FreeDS esclavo se comporta como un FreeDS esclavo, Pablo tiene que cambiar menos cosas, y todos tan contentos.

Aquí hay flexibilidad (dependiendo de cuánto tiempo le pueda dedicar yo).

E insisto: por favor, no deis por hecho que HA implica MQTT.

pablozg, 2023-05-28:
No me importa añadir una nueva fuente de datos que se conecte por websockets a HA, solo me tienes que indicar que datos recibirá y a que corresponde cada uno, si no requiere mucho tiempo, puedo tratar de añadirlo para la beta 18.

IvanSanchez, 2023-05-29:
Pues he levantado un HASS de pruebas en `http://ivan.sanchezortega.es:18123`, y supongo que lo dejaré funcionando unas cuantas semanas. La interfaz de websockets está en `ws://ivan.sanchezortega.es:18123/api/websocket` . Tiene un usuario `pablozg`.

Primero hay que obtener el token de autenticación de Oauth2. El cómo obtener el token está explicado en https://developers.home-assistant.io/docs/auth_api/ , y se resume en:

- La página web de FreeDS genera un enlace como
  ```
  http://direccion-de-hass/auth/authorize?
    client_id=http%3A%2F%2Fdireccion-de-freeds&
    redirect_uri=http%3A%2F%2Fdireccion-de-freeds%2Foauth2_callback
  ```
- El usuario sigue el enlace, y HASS hace cosas internamente.
- HASS hace un HTTP POST a FreeDS, en http://direccion-de-freeds/oauth2-callback, conteniendo un JSON como:
  ```
  {
    "access_token": "ABCDEFGH",
    "expires_in": 1800,
    "refresh_token": "IJKLMNOPQRST",
    "token_type": "Bearer"
  }
  ```
- FreeDS se guarda el token de acceso y de refresco en flash.

Una vez que FreeDS tiene un token oauth2, puede conectarse al flujo de websockets.

El primer mensaje de websockets es desde el FreeDS hacia el HASS, mandando el token de autenticación:

```
{"type":"auth","access_token":"ABCDEFGH"}
```

A lo que HASS responde con:

```
{"type":"auth_ok","ha_version":"2023.5.4"}
```

Después FreeDS le dice a HASS que quiere enterarse cuando cambie el estado de cualquier cosa:

```
{"type":"subscribe_entities","id":1}
```

A partir de ahí, HASS empieza a mandar mensajes con los valores de todo lo que cambie. Un mensaje típico tiene esta pinta:
```
[
  {
    "id": 1,
    "type": "event",
    "event": {
      "c": {
        "sensor.fake_grid_power": {
          "+": {
            "s": "-362.1",
            "lc": 1685367048.816932,
            "c": "01H1KV8JKGME4KEV6JWN4C27J0"
          }
        }
      }
    }
  },
  {
    "id": 1,
    "type": "event",
    "event": {
      "c": {
        "sensor.fake_solar_power": {
          "+": {
            "s": "1629.5",
            "lc": 1685367048.817347,
            "c": "01H1KV8JKHBAX5JXRXFM6RKAAN"
          }
        }
      }
    }
  }
]
```

pablozg, 2023-05-29:
En estos momento ando liado, si tengo algún hueco le daré un vistazo, lo más importante indicar que los nombres de los sensores deben cumplir una serie de requsitos y que tenga el nombre fijo en cualquier integración, ya que poder hacerlo personalizable me supondria mucho más tiempo y una modificación sustancial del código actual, porque el hecho de solicitar el token y almacenarlo no supone problema, así como realizar la comunicación.

IvanSanchez, 2023-05-29:
> lo más importante indicar que los nombres de los sensores deben cumplir una serie de requisitos y que tenga el nombre fijo en cualquier integración

No, no es posible. Lamentablemente el administrador de HASS puede cambiar los nombres de cualquier sensor.