pablozg/freeds #88: Residual power no modo automático

brunofcat, 2023-04-27:
Olá.

O meu termoacumulador tem um display com consumo constante.
Existe forma de, no modo automático, manter uma carga residual em vez de desligar por completo?

Obrigado.

brunofcat, 2023-05-02:
Olá,
Deixo aqui o modelo do meu termoacumulador que tem um display:
https://www.fleck.pt/fitxa_producte.asp?id=29

Obrigado.

pablozg, 2023-05-03:
Hola. en es posible dejar una salida constante, ya que el regulador trabaja recortando la onda y no es recomendable usarlo con termos que lleven electronica, en todo caso podrías comprobar si la resistencia del termo lleva protección mecánica, y si es así podrías conectar el freeds a la resistencia directamente en lugar de al termo. 

Creo recordar que en el manual que puedes encontrar en el grupo de telegram se incluyo una modificación para hacerlo funcionar con termos que llevan electronica.

brunofcat, 2023-05-05:
Hola Pablozg,

Ficou claro o modo de funcionamento do regulador e dessa forma não faz sentido ter uma carga residual.

O meu termoacumulador tem 2 compartimetos, ou seja, 2 resistencias e 2 sensores de temperatura.
Já vi um esquema no grupo de telegram com 2 reguladores no entanto estou a ponderar usar contactor.

Vou continuar este tópico no grupo de Telegram visto que a solução não passa pelo software.

Obrigado.

closed 2023-05-05 by brunofcat