pablozg/freeds #92: Error en punto decimal de vertido en vista Monitorización/Detalles

rompesf, 2023-06-02:
1.1.0017 Beta Build: 22052023 Web: 1.1.3

Hola, actualicé recientemente a esta versión y en la vista de detalles de monitorización aparecen algunos valores erróneos porque la coma decimas está desplazada un dígito.
[image]

El Vertido(Ayer) debería ser 15.4 en vez de 1.54 y lo consumido hoy también muestra un valor erróneo.

rompesf, 2023-06-02:
Por cierto, con inversor Goodwe EH

pablozg, 2023-06-02:
Debe ser un problema en tu instalación, freeds se limita a calcular el vertido / consumo según los datos que recibe revisa la velocidad de refresco de los datos y / o si has sufrido algún reinicio, los datos solo se guardan a las 00:00 de la noche para no degastar la memoria interna, si sufres un reinicio, los datos calculados se pierden.

Aquí tienes una muestra de lo que mide mi freeDS con Shelly como fuente y el propio shelly en su nube:

[image]

closed 2023-06-02 by pablozg