pablozg/freeds #96: ICC Solar (MQTT) / Error en la recepción de los datos de medida

cosygarcia, 2023-09-01:
He instalado recientemente el gestor freeds (hardware: gen9 / software: 1.1.0014 Beta BETA Build: 22112022).

Mi inversor (Solax X1 mini 3k) y mi panel eléctrico principal están alejados uno del otro, y no puedo tener un solo Shelly que mida la potencia solar generada y la potencia de la red. Transmito la potencia solar generada a Home Assistant consultando las API locales del inversor. Transmito la potencia de la red a Home Assistant a través del sensor Shelly 3EM.

He creado 2 triggers en Home Assistant:
- Uno que, cada vez que cambia el valor de la potencia solar generada, publica ese valor en mi broker MQTT en el topic "Inverter/PvWattsTotal".
- otro que, cada vez que cambia el valor de la potencia de la red, publica ese valor en mi broker MQTT en el topic "Inverter/GridWatts".
- 
Finalmente, he configurado el gestor freeDS para utilizar una fuente de datos de tipo "ICC Solar (MQTT)". He configurado la conexión MQTT en freeDS.

La página de Monitorización muestra que el excedente se deriva hacia el termo durante unos segundos. Luego, obtengo el mensaje de error "Error en la recepción de los datos de medida" en la página de Monitorización. Al ir a la página de la consola/weblog, tengo los siguientes mensajes de error:
16:59:25 - PWM: Apagando PWM por superar el tiempo máximo en la variación de los datos del consumo de Red
16:59:58 - PWM: Apagando PWM por superar el tiempo máximo en la variación de los datos del consumo de Red
17:00:31 - PWM: Apagando PWM por superar el tiempo máximo en la variación de los datos del consumo de Red
17:01:04 - PWM: Apagando PWM por superar el tiempo máximo en la variación de los datos del consumo de Red

Cual es el problema ? Gracias de antemano

pablozg, 2023-09-06:
El problema es que el dato que recibe como valor de red no varia dentro del tiempo de espera antes de dar error, por defecto son 20 segundos, eso quiere decir que no varia su valor desde el ultimo recibido en ese margen de tiempo, aumenta ese valor, o modifica si puedes para que envie el valor de forma periodica con una pequeña variación, puede ser 0.01 a 0.09 watios de forma aleatoria si quieres para no influir en la decisión que toma freeds pero es suficiente para detectar un cambio en su valor.

cosygarcia, 2023-09-06:
He descubierto que Home Assistant solo recopila los valores del Shelly cada
30 segundos. Así que he creado un nuevo sensor "rest" que se actualiza cada
2 segundos. Parece funcionar mejor ahora.

articoweb, 2023-09-14:
@cosygarcia Una consulta, qué es eso de hardware Gen9? Sólo vi placas de solarhomer v1 y v2. Soy nuevo, disculpar que haga tan patente mi ignorancia.

pablozg, 2024-01-12:
> @cosygarcia Una consulta, qué es eso de hardware Gen9? Sólo vi placas de solarhomer v1 y v2. Soy nuevo, disculpar que haga tan patente mi ignorancia.

Siento haber leido tan tarde tu pregunta, el hardware Gen9 es el que suministra Francisco de Solares TV, decir que no tiene nada que ver con freeDS, tan solo es un miembro más que facilita hardware compatible tal y como hace solarhomer y otro usuarios, informate bien de los pros y contras de cada placa ya que existen varias versiones que usan tecnologia difente. 

Puedes pasarte por el grupo de telegram si quieres estar más informado.

closed 2024-04-03 by pablozg