pablozg/freeds #101: CONTROLAR SALIDA POR PRODUCCION SOLAR (PID Setpoint tracking)

isabido, 2024-01-03:
Buenas @pablozg estoy tratando de conseguir un funcionamiento un tanto especial.

Tengo una produccion solar de solo 2Kw, y quiero priorizar todo la generacion solar para calentar el agua. Para evitar de usar el Butano. 

Para eso necesitariamos que la entrada  del PID fuera el valor de PRODUCCION SOLAR y no el GRID.

[image]

En este caso, seria un un funcionamiendo de un PID en modo setpoint tracking, es decir el SP seria el valor de PRODUCCION Solar.

Ejemplo si los paneles estan generando 1KW, la salida tendria que ajustarse para generar un consumo de 1KW, (lo ideal seria hacerlo tomando la medida de un CLAMP para que fuera lo mas real posible, pero bueno se podria hacer como esta ahora mismo FREEDS con valor calculado sobre la salida PWM.

A ver que te parece!

[image]

pablozg, 2024-01-09:
Lo que pides no tiene sentido por lo menos para mí, si estás produciendo 1kw y solo usas el valor de producción, siempre usaras energía de red eléctrica (gastaras el equivalente a los consumos que tengas en casa), la gracia del sistema de freeds es poder usar la energia disponible sin gastar energia de la red.

Por otra parte el calculo que hace del consumo no es exacto y no sería válido para mantener como consigna, si quieres conocer cuando el termo está caliente puedes usar la opción de la pinza amperimetrica que ya soporta freeds y así cuando el consumo sea 0 y el pwm 100% activar el limitador que quieres usar.

Tambien puedes usar la opción de paso a modo manual según producción, por defecto si el valor de producción es inferior al valor que configuras se pasa a modo manual de forma automatica, pero tienes un comando 'invertPVPowerLogic' que permite invertir la lógica y hacer que pase a modo manual si supera cierta producción, y tal vez te sirva para lo que quieres.

Pero por el momento no veo sentido ni utilidad general en implementar lo que quieres más allá de tu propia necesidad, lo siento, puedes preguntar en el grupo de telegram que soluciones usan otros miembros para lograr algo parecido, tal vez tengas suerte y encuentres a alguien que pueda darte otra solución.

Saludos.

isabido, 2024-01-09:
Gracias Pablo, entiendo que te pueda parecer raro, pero en una instalación donde solo hay 2kw de maxima produccion y la prioridad es calentar el agua, es casi imposible hacerlo SOLO con el excedente. Ya que esa resistencia con menos de 500W ni se entera.

Ahora mismo lo tengo solucionado con NODERED y un PID donde uso de SP la produccion solar (por MQTT desde el inversor), PV valor medido Shelly 1PM en la resistencia del termo, y un dimmer 0-10v shelly conectado a un SSVR. 

Con esto voy volcando todo la produccion solar al termo, y cuando llega a 60º grados paso a modo excedendes hasta los 70º.

De esta forma logro calentar el agua como prioridad. Si lo maduras un poco, creo que le encontrarias el sentido. Efectivamente la casa seguira consumiendo "lo de siempre" durante el calentamiento del agua, pero la idea es no usar butano al final del dia. 

En modo excedentes, como va sobrando muy poco al final del día no llego a calentar el agua lo suficiente como para ducharse y he tirado toda esa energia para al final tener que arrancar el butano.