pablozg/freeds #102: Lost Config

Glammies, 2024-01-08:
Hi

Sometimes my esp with [Release 1.0.7 Final Rev 2] reset all configs. Several times in a month other times only after 2 ou 3 months. 
Like was the first time connection.
 Several times in a month other times only after 2 ou 3 months. 
I don´t know what is the problem. Coul you please help?
Thanks

pablozg, 2024-01-09:
Hi, your version is outdated and not supported, you must use the beta 18 o above in other to get support.

Thanks.

Glammies, 2024-01-23:
Thanks my friend
Already update for the beta version 18 and so far so good
Could you please update link for telegram? I want learn more about freeds but the telegram link is not working.
Sorry for the late response

closed 2024-04-03 by pablozg