pablozg/freeds #103: ICC Solar (MQTT)

egomes951, 2024-01-19:
Hola,
Estoy usando la version 1.0.7 rev 2 y no logro controlar el PWM (ON, OFF) asi como tampoco consigo controlar el PWM Manual.
Es decir, para controlar el PWM 
    - Estoy enviando la carga {"command":"pwm", "payload":"0"}  para apagar y {"command":"pwm", "payload":"1"}  para enceder al topico freeds_bd68/cmnd/pwm.
    - Resultado, sea cual sea el payload siempre recibo el output 10:31:23 - Mqtt - PWM control: OFF 10:31:23 - DATA SAVED!!!!

Para controlar el modo:
   - Estoy enviando la carga {"command":"pwmman", "payload":"0"}  para automatico y {"command":"pwmman", "payload":"1"}  para manual al topico freeds_bd68/cmnd/pwmman
   - Resultado: sea cual sea el payload siempre recibo: 10:34:09 - Mqtt - PWM mode set to: AUTO 10:34:09 - DATA SAVED!!!!

pablozg, 2024-01-20:
La version 1.0.7 rev 2 ya está obsoleta y no se le presta soporte, además estás intentando usar la forma de enviar comandos que se usa en las betas, por lo que te sugiero que actualices el esp32 con el bat incluido en la última beta, en estos momentos las 18, aunque la 19 no tardará en publicarse.

egomes951, 2024-01-21:
Hola buenos días, Actualizado a la ultima version beta y funciona!! :)
Topico:  freeds_bd68/cmnd
Carga:   {"command":"pwm", "payload":"0"} --> para apagar  {"command":"pwm", "payload":"1"} --> para enceder.

Gracias,
Elisio

closed 2024-01-21 by egomes951