pablozg/freeds #105: question: does FreeDS support a zero export system

geo99918, 2024-02-21:
Hi,

for some reason I could not join the Telegram channel thus I'm posting here.
The question is : does FreeDS work when the inverter (SUN2000-6KTL-L1) is set up for zero export ?

thank you
Geo

pablozg, 2024-02-21:
Hola,

Dime tu nick de telegram y comprobaré si está baneado, los baneos occurren
si te inscribes y no escribes un mensaje de presentación.

Huawei es especial en su configuración, pero si puede usarse con vertido
cero, solo tendrás que permitir que freeds consuma algo de energia de la
red para que de tiempo al inversor a regular la producción, cuando entres
en el grupo lee el manual que encontrarás en el mensaje de bienvenida,
cualquier duda tras su lectura, preguntala en el grupo.

Saludos.

geo99918, 2024-02-21:
telegram id: "geo 000" , I think, not using telegram frequently
thanks for the answer